prusa3d/PrusaSlicer #13058: PrusaSlicer 2.8.0 configuration wizard auto adds SLA printer

kassi, 2024-07-09:
### Description of the bug

When I open the configuration wizard (after having set up everything already a long time ago), the Original Prusa SL1S SPEED printer is added (checked).
I realized that when adding filaments and closing the wizard, that the printer changed from my MK4 to a (non existent) SL1S.
If I open again, it's added again and I always have to uncheck it.

### Project file & How to reproduce

[EmptyProject.3mf.zip](https://github.com/user-attachments/files/16151203/EmptyProject.3mf.zip)
- Check currently available printers (should not show SLA)
- Open wizard
- Close wizard
- => Printer changes to SL1S
- Open wizard
- Choose SLA printers
- uncheck SL1A
- Close wizard
- => Printer shows MK4, not SLS visible
- OPen wizard
- Choose SLA
- => see that SL1S is again checked

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-arm64

### Operating system

MacOS

### Printer model

MK4+MMU3

YuSanka, 2024-07-17:
Thanks for your report.
SPE-2419

WildRikku, 2024-08-09:
The MK4 is also automatically added, which the OP probably didn't notice because they actually use an MK4. I use a Creality printer but the two Prusa printers are added every time I open and close the configuration assistent.

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena