prusa3d/PrusaSlicer #13059: Performance issues, Edit Window, OSX 10.14, MBP17 2010

DerKameramann, 2024-07-10:
### Description of the bug

2.8.0 GUI performance in Edit is sometimes retarded, Preview is barely responding at all and shows nothing onto the printbed.

Hardware had no problems at all since 11/2020 up to 2.7.4 (going back to 2.7.4 causes issues too)

### Project file & How to reproduce

[mcneill_Flaschenhalshalterrakete.3mf.zip](https://github.com/user-attachments/files/16158203/mcneill_Flaschenhalshalterrakete.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-x64

### Operating system

OSX 10.14.6

### Printer model

Prusa MK3s

DerKameramann, 2024-07-10:
[image]

MacUrb, 2024-07-10:
Have experienced the same. I was trying to define a color change at certain heights. 
a) the slider is not responding anymore as it did in 2.7.x. I reverted back and haven't seen any issue - so the issue is not related to the hardware.
b) the slider itself has changed such that the print-time 'bubble' is hiding some (super small) items on the slider
c) I'm missing the slider which is showing the nozzle movement

DerKameramann, 2024-07-10:
i think i´ts an hardware issue - somewhat - our trusty old hardware seems to fade out. the compiling processes drop old communication channels and thats an issue to the old nvidia grafics because apple went to amd long ago.
i failed going back to 2.7.4 yet. try it again later...

kubispe1, 2024-07-10:
Please check this issue: #12908. 
This may be a duplicate. Please discuss your issue there.

DerKameramann, 2024-07-10:
OK, thanks. Just want to point out, #12908 is Windows related not Mac but yes, same sort of issue.
[image]

kubispe1, 2024-07-11:
OK, thanks. We will need to handle it.
I am closing this issue.
Thanks

closed 2024-07-11 by kubispe1