prusa3d/PrusaSlicer #13062: Layer height slider issue

IonNight, 2024-07-10:
### Description of the bug

Hi I just uploaded to PrusaSlicer-2.8.0

The slider works fine as is, but if you hold the mouse and drag the slider to the top and while still holding down mouse right, drag it over "Print settings:" and "Filament:" and down over that panel and a white transparent elongated box will appear behind the mouse pointer and the layer height slider dont work anymore with that click of the mouse.

https://github.com/prusa3d/PrusaSlicer/assets/27898087/f728e676-9f5e-4546-a085-1c8d8df5ef21

 

### Project file & How to reproduce

This is the model in the project file: right click -> add shape -> box

The stl/3mf is not the issue here
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10 Pro 64-bit 

### Printer model

mk3s

YuSanka, 2024-07-17:
Thanks for your repost.
SPE-2418

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena