prusa3d/PrusaSlicer #13066: 2.8 Seams poor alignment on some models without sharp corners

danukefl, 2024-07-10:
### Description of the bug

On version 2.8.0, the seam placement on prints with a radius are not detected as a suitable location for seams when no sharp corners are present and inherits from the surrounding features into the closest location to those surrounding features. 

I understand that this is the expected result for a more consistent seam but the seam placement appears "too selective" on the location. This doesn't work well cosmetically with some models as it is common to add a small radius to sharp corners to enhance printability and speed. Another side effect is that seams can be placed into a smooth surface with higher speeds at that point deteriorating the surface quality. Painting the seam does work as expected but is a time consuming workaround when even a 1mm radius is not used for a corner and I personally would just continue to the older version.

I created a model (attached) to highlight this as it goes from a square with sharp corners to another with radius corners ranging from 1mm-4mm. In 2.8 the seam follows the sharp corner all the way to the top of the model, but in <=2.7, the seam goes to the smallest radius.

v2.8
[image]

v2.7
[image]

Having the seam selection algorithm be more "relaxed" in the corner selection would resolve this. A way to adjust this sensitivity would be a great option if possible to fit more peoples' needs. 

### Project file & How to reproduce

[model.zip](https://github.com/user-attachments/files/16168781/model.zip)

Load and slice both models into PS 2.8 and 2.7 with aligned seam position, and enable seams in the preview. Print and filament settings do not make a difference.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

XL w/ IS and MK3S+

SachCZ, 2024-07-11:
Hello @danukefl, thank you for the time and investigation. It is not easy to determine when the seam placing algorithm should choose a round corner over continuing the "aligned" path. In some cases one is better than the other and vice versa.
I have hard time imagining a solution that would **work great for all the possible models** without any fine tuning on the user side (for example the one you describe). 

Currently we are more inclined to leave the fine tuning to the painting as we strongly believe that it is best to keep things **less complex** even though sometimes bit more time consuming for the end user. To loosely paraphrase what I am saying: "You cannot have a button for everything" and "There should preferable be just one way to do things". My **personal** opinion is that what is lacking at the moment is the painting itself, which can be a bit cumbersome to use sometimes, thus leading to the end user frustration.

It seems to me that what you are describing is a **feature request** to have more control over when the seam placing algorithm picks up a corner. I believe so based on the examples you provided as the seam placing algorithm did not fail spectacularly, just **did not produce the best possible result**. I am thus labeling the issue as such.

All this does not mean we will not implement implement the solution you are proposing or that we will not fine tune the algorithm in future release.

neophyl, 2024-07-11:
I'd have to agree with SachCZ, I looked at that test example and thought that 2.8 placed the seam better and more logically when left to its own devices.  

You can always use seam painting to get more precise placement in specific areas when you want something different to the algorithm.

danukefl, 2024-07-11:
@SachCZ I understand and do agree that the placement is better for most models, selling personalized models and the main one being severely impacted by the changes and with a lacking seam painting tool make this frustrating and time consuming for myself which is a direct cost, if this was purely a hobby, I wouldn't care because overall I am happy with the changes and would not want it reverted since it is better for everyone else probably. 

I've attached a photo showing the difference before and after and how bad the seams came out after updating on one of my models. Without changing my profiles, it creates nearly a 2mm gap (print setting specific that I would have to tune if I were to leave the seam there) in the face as the seam follows the lower V arm straight up through the flat face, versus on the old model where the seam goes from the V arm to the edge. I created the test object to highlight the difference specifically.

[image]

I can see many options for this in addition to just tuning the algorithm.

- Select Legacy versus New algorithm like with the classic versus arachne perimeters
- A "threshold" option to adjust the algorithm on rounded corner sensitivity
- Additional options for the seam placement Corner/Pretty/Hidden in addition to the Rear/Aligned/Etc.

In regards to seam painting, my largest gripe is that it is entirely freehand. My suggestion is to implement a few basic tools like draw a straight line between two points at a minimum, or to implement a way to select lines/features similar to the Measure tool. I do agree the seam painting enhancement would probably be the best way to go for future development for everyone.

neophyl, 2024-07-11:
For painting have a read of https://github.com/prusa3d/PrusaSlicer/issues/12775#issuecomment-2146111648  (if you aren't on a mac)
Not as good as a proper tool which has been requested multiple times but a handy work around in many cases.

Theres another workaround in this request https://github.com/prusa3d/PrusaSlicer/issues/7059#issuecomment-1436049545

SachCZ, 2024-07-12:
Hello @danukefl, in the picture you provided it looks almost suspicious that the corner is not evaluated as sharp corner and I can feel your pain :smiling_face_with_tear:.

We will look into tuning the parameter responsible for it. I created an internal issue for it: SPE-2412.

danukefl, 2024-07-12:
Thanks @SachCZ. Just for your notes, that corner has a radius of 0.05" (1.27mm) designed in Fusion 360. If I remove that radius, is detected properly as a corner, but for my use case I can't have a sharp corner there,

brigthub, 2024-07-17:
I am seeing printing z-seam problems in general with version 2.8.  Not a z-seam to be found on part printed from version 2.7.1 left side of picture.  The right side part has a huge z-seam line.  Same printer and same filament.
[image]

charris3621, 2024-08-18:
I am seeing printing z-seam problems in general with version 2.8. Not a z-seam to be found on part printed from version 2.7.1 left side of picture. The right side part has a huge z-seam line. Same printer and same filament. - I copied Brigthub because am seeing the same thing with my recent prints.
[image]

danukefl, 2024-08-18:
@brigthub @charris3621, try it with the default Prusa profiles to see if it goes away or is different at least. For me after upgrading, I had to compare my custom profiles to see what changed or I was having massive gaps in the seam. Because of the placement issues, I still use 2.7.x though.

charris3621, 2024-08-19:
Thank you, for your help. I will check to make sure I am using the default Prusa profile. If the problem persists, I will have to learn how to revert back 2.7. I had just started my first 3d print order from someone who asked for 12 prints. Such bad timing.

danukefl, 2024-11-21:
@SachCZ I just downloaded the 2.9.0 alpha and it looks like the seam placement is playing nicely on the radius corners again. Thanks for getting this submitted and to anyone that was involved in the tweaks required.

[image]

danukefl, 2024-11-22:
Bad news and I hate to reopen it @SachCZ, I was able to recreate it by just changing my layer height from 0.3mm to 0.2mm on my same custom print profile but also with the built in 0.20mm speed/quality profile when changing it from 0.20mm to 0,30mm.

Digging in some more, playing with 0.20mm SPEED on the Mk3s+ OOB profiles, if I change only the layer height and nothing else, from 0.2mm to 0.22mm layer heights exactly, the seam comes up the flat side from the lower arm which is not desired. If the layer height is 0.199mm or less, or 0.221mm or more, then it goes to the edge as desired. I made a table of the layer heights I tested and the result. My hunch is some funky math around the 0.2mm layer height, or some path optimization that is only occurring in the range, but I have no clue as I'm not versed in the code.

| Layer Height | Seam Result |
| ---- | ---- |
| 0.1mm | Edge |
| 0.15mm | Edge |
| 0.18mm | Edge |
| 0.19mm | Edge |
| 0.199mm | Edge |
| 0.2mm | Flat Side |
| 0.21mm | Flat Side |
| 0.22mm | Flat Side |
| 0.221 | Edge |
| 0.23mm | Edge |
| 0.25mm | Edge |
| 0.3mm | Edge |

closed 2026-09-29 by prusaslicer-bot[bot]