prusa3d/PrusaSlicer #13068: Exclude Wipe Tower from Arrange

MysticGringo, 2024-07-11:
Please exclude the wipe tower from being arranged when using the arrange button. I prefer the wipe tower to be in it's default upper right corner, or where I specifically put it, and wish it would be "locked" in position so when arranging parts it doesn't move.

PrusaSlicer 2.8.0. MK4 MMU Profile.

lcemoore, 2024-07-13:
I too prefer to leave wipe tower in the corner and I have to move it every time I use the arrange feature.  It may be silly, but this extra step has made me avoid using the auto arrange feature.

elxris, 2024-07-24:
In the meantime you can do `Ctrl + A` (Select all) and `Shift + A` (Arrange selected). The wipe tower will not be selected with Select All command.

MysticGringo, 2024-07-26:
> In the meantime you can do `Ctrl + A` (Select all) and `Shift + A` (Arrange selected). The wipe tower will not be selected with Select All command.

That sort of works... it arranges all my parts **around** the wipe tower. Granted, it doesn't move the tower, just all my parts to be up around it.

ex-nerd, 2024-08-30:
See also: #11785

ex-nerd, 2026-09-18:
See update to https://github.com/prusa3d/PrusaSlicer/issues/11785#issuecomment-5728675407

With v3 there doesn't (yet?) seem to be any way to change `Arrange` to specify the anchor point when the purge tower exists, which means the purge tower (and parts) are always moved to the rear-center of the print bed. Until that gets fixed, this particular issue can't even be tested against v3.