prusa3d/PrusaSlicer #13070: Wipe Tower prints final wipe in mid-air

MysticGringo, 2024-07-11:
### Description of the bug

In certain situations, such as creating "label" style prints, the final purge of the filament on the wipe tower is in mid-air. I presume this is done to create the proper tip for the filament, however it leaves a string of filament that hangs to nozzle or falls onto or behind plate.

[image]

[image]

It's repeatable, and consistent.

### Project file & How to reproduce

[ZipperPulls.3mf.zip](https://github.com/user-attachments/files/16177653/ZipperPulls.3mf.zip)

Using any profile with wipe tower and a label style print, it is easily reproduced. By label style print, I mean a 2 color print with 1 color only being used for the first few layers (2-4) while rest of print is single color and thus no wipe tower gets built. Turning on "no sparse layers" does not change outcome.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Mac

### Printer model

MK4 MMU3

k1-801, 2024-07-29:
Can confirm, also experiencing this on some models.

neophyl, 2024-07-29:
Please see https://github.com/prusa3d/PrusaSlicer/issues/11904

DandanLam, 2024-08-26:
I was able to reproduce this in PrusaSlicer 2.7.4 as well as 2.8.0; both running on macOS for MK3S+ MMU3

k1-801, 2024-08-26:
For those who didn't open the linked issue:
the extrusion in the air is in fact NOT a bug, it is a misrepresented part of the tip forming procedure (ramming), where a short length of filament is extruded before the filament is removed from the nozzle completely. It might not even come out of the nozzle, despite being shown on the preview.