prusa3d/PrusaSlicer #13072: "Nearest" seam position not working properly, inconsistently ignores corner positions

3dPrinting1127, 2024-07-11:
### Description of the bug

This might be me misinterpreting how the "nearest" parameter optimizes the seam position, but I thought that this function places the seam position at the nearest corner with respect to the last movement on the previous layer.  Here's what I'm seeing on my print:

[image]

As you can see, the seam position isn't always being placed at an obvious nearby corner on the model.  This is resulting in those small zits appearing.  For the unwanted seam position that's not in a corner, here's the last movement for the previous layer:

[image]

The unwanted seam position seems unnecessary as there is an obvious corner (among others) nearby, so I'm not sure why this is happening.

### Project file & How to reproduce

[test model.zip](https://github.com/user-attachments/files/16180798/test.model.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusaslicer 2.8

### Operating system

Windows 11

### Printer model

MK3S+

SachCZ, 2024-07-12:
Hello @3dPrinting1127, I have confirmed your issue. It seems to be a bug. I have created an internal issue: SPE-2413.

robustini, 2024-07-15:
Confirmed, those set as nearest are totally ignored.

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena