prusa3d/PrusaSlicer #13074: Organic supports ignore object boundary and create them inside other objects.

TheApsik, 2024-07-11:
### Description of the bug

Organic support are created inside other object. PrusaSlicer detect this issue and shows error:
[image]
[image]

### Project file & How to reproduce

Open error.3mf file and slice it.
[error.zip](https://github.com/user-attachments/files/16181165/error.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 10

### Printer model

Voron 2.4r2

TheApsik, 2024-07-16:
Any actions? Update? Everyone on vacations?

neophyl, 2024-07-16:
Maybe because this is duplicate of many many previously raised issues.  Its not just organic supports, its ANY supports.  Actually its ANY object too if you decide to place them overlapping.  Its intended operation/design and has been that way forever and it doesn't look like its being changed so get used to it.  

Be thankful you now get a warning, that's relatively recent.

Things are sliced individually, its what allows you to set different print parameters for things.  If you want to stop it happening then use Merge object once you have things placed and then it wont put support through.  Of course then you lose the ability to print differently for different objects.

https://github.com/prusa3d/PrusaSlicer/issues/10464
https://github.com/prusa3d/PrusaSlicer/issues/10451
https://github.com/prusa3d/PrusaSlicer/issues/10376
https://github.com/prusa3d/PrusaSlicer/issues/10032
https://github.com/prusa3d/PrusaSlicer/issues/9861
https://github.com/prusa3d/PrusaSlicer/issues/9860
https://github.com/prusa3d/PrusaSlicer/issues/9835
https://github.com/prusa3d/PrusaSlicer/issues/9685
https://github.com/prusa3d/PrusaSlicer/issues/9517
https://github.com/prusa3d/PrusaSlicer/issues/9510
https://github.com/prusa3d/PrusaSlicer/issues/8012

I could go on with the list.  A cursory search using support through would have found all of those plus more.

TheApsik, 2024-07-16:
Wow.. +2 years to find solution.. OK, whatever..

closed 2024-07-16 by TheApsik