prusa3d/PrusaSlicer #13075: Outside perimeter print first on complex overhang *sometimes

thatguymendel, 2024-07-11:
### Description of the bug

   This happens when two chamfered holes intersect.

This is the model in fusion with a 45 degree chamfer
[image]
This is what it looks like in prusaslicer
[image]

This is the preview showing the outer perimeter printing first
[image]
This is especially not good with very steep overhangs, where the perimeter will have a chance to print properly by sticking to the perimeter behind it, but currently it will print in mid air

This is chamfered 70 degrees
[image]

when i made the 70 degree model, the first time i sliced it the outside perimeter printed in the order its supposed to. I adjusted the spacing of the holes slightly and i was able to replicate the issue. 

[image]

this is the 45 degree chamfer model, and all i did was add a cube in prusaslicer to lift it off the ground to see if its a problem with just the second layer. 
[image]
this is the exact same model but now it prints fine

I tried changing the fist layer height from 0.3 to 0.2 and that seemed to fix the issue, but i was able to fiddle with the model to replicate the problem with a 0.2 first layer height

### Project file & How to reproduce

These are all the models that are in the pictures
They are very simple cubes with holes cut out of them and the bottom of the holes are chamfered to produce an overhang
[Test Models.zip](https://github.com/user-attachments/files/16183007/Test.Models.zip)

layer height = 0.2
first layer height = 0.2/0.25/0.3
perimeters  = 3

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Sidewinder X1

agrif, 2024-07-29:
I'm just commenting here to note that this issue looks [very similar to what I'm seeing](https://github.com/prusa3d/PrusaSlicer/issues/13144#issuecomment-2255297363) and may be the same as #13144.

thatguymendel, 2026-09-03:
Still happens in prusaslicer 3.0.0