prusa3d/PrusaSlicer #13076: Graphics performance very bad with PrusaSlicer 2.8.0

the-powl, 2024-07-11:
### Description of the bug

The graphics performance of PrusaSlicer 2.8.0 seems to have largely declined on my machine. I'm using the internalt Intel UHD Graphics 770 of my Intel Core i5 Processor and that's usually sufficient. But with the newest version of PrusaSlicer when I want to rotate the view on a half decent model in the sliced Preview the framerate drops to below <1fps. For large prints this is not even usable any more.

### Project file & How to reproduce

gdsgf

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

irrelevant

kubispe1, 2024-07-12:
Hi, please attach your system info to provide us with more data. It may help us.
If you export the G-code and then open it in a G-code viewer, is the issue observed as well?
[image]

writer-sol, 2024-07-14:
Hello, same problem. After opening G-code in G-code viewer, the problem is also observed.
PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (сборка 19045), 64-бит редакция
Total RAM size [MB]: 17,124MB
OpenGL installation
GL version:   3.3.0 (3.3.0)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     GeForce 8800 GT/PCIe/SSE2
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_clear_buffer_object
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robust_buffer_access_behavior
[... 162 lines cut ...]
GL_NV_texture_env_combine4
GL_NV_texture_expand_normal
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

foxylion, 2024-07-20:
The same for me. Previous versions where good to use. The 2.8.0 is unusable in G-Code preview (I can't rotate the screen anymore for big models).

Specs:
```
PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 17,109MB
OpenGL installation
GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2)
Profile:      Core
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 128 bits)
GLSL version: 3.30.0
Textures compression:       Disabled
Mutisampling: Disabled
```
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_sRGB
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_shader_integer_mix
GL_EXT_texture_array
GL_EXT_texture_compression_rgtc
GL_EXT_texture_integer
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_ycbcr_texture
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_OES_EGL_image
GL_OES_read_format
</details>

DerKameramann, 2024-07-24:
Same here with Apple OSX Mojave on i7MBP17/2010
-----------------------------------------------------
PrusaSlicer
Version:   2.8.0+1-lm-spe2411-MacOS-x64
Build:     PrusaSlicer-2.8.0+1-lm-spe2411-MacOS-x64-g86e8e256a-202407171406

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Mojave Version 10.14.6 (Build 18G103)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version:   3.3.0 (3.3 NVIDIA-10.4.14 310.90.30.05b27)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_explicit_attrib_location
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shading_language_include
GL_ARB_texture_cube_map_array
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_texture_mirror_once
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

k1-801, 2024-07-31:
Same on my machine.
Although I don't see any performance drops in the slicer, I see it affecting other graphics-heavy applications when the slicer is in the background, doing nothing and showing a static picture.
Task manager shows the slicer running my GPU at 75% - and that's for an NVidia 4060.

```PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (сборка 19045), 64-бит редакция
Total RAM size [MB]: 16,781MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 556.12)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 4060 Laptop GPU/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
```

<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
[... 333 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

trashmaster9000, 2024-08-02:
same here, Model view is working without any issue even with models over 1M tris, gcode view is barely usable.

```
PrusaSlicer
Version:   2.8.0+linux-x64-GTK3
Build:     PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.1.0-23-amd64 x86_64
Total RAM size [MB]: 7,202MB
OpenGL installation
GL version:   4.6.0 (4.6 (Core Profile) Mesa 22.3.6)
Profile:      Core
Vendor:       AMD
Renderer:     AMD Radeon Vega 8 Graphics (raven, LLVM 15.0.6, DRM 3.49, 6.1.0-23-amd64)
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
```
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
[... 161 lines cut ...]
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_NVX_gpu_memory_info
GL_NV_alpha_to_coverage_dither_control
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_NV_shader_atomic_int64
GL_NV_texture_barrier
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_S3_s3tc
</details>

the-powl, 2024-08-06:
Here's mine:

PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 34,141MB
OpenGL installation
GL version:   4.6.0 (4.6.0 - Build 31.0.101.3959)
Profile:      Core
Vendor:       Intel
Renderer:     Intel(R) UHD Graphics 770
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
[... 170 lines cut ...]
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_OVR_multiview
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

k1-801, 2024-08-19:
[image]

89% GPU load when it is showing a static image.

ChiefPoints, 2024-09-02:
I'm seeing the exact same thing on MacOS Monterey, PrusaSlicer 2.8.0 is completely unusable when in the sliced gcode view, moving rotating the object, view layers and just clicking in the interface produces horrible lag.  In addition to this here, I've relayed the same info in email to prusa. Version 2.7.4 was rock solid, no issues. Really need this working so I don't have to manually build profiles for the MK4S in version 2.7.4 - Here's the system info:

Version: 2.8.0+MacOS-x64
Build: PrusaSlicer-2.8.0+MacOS-x64-202406270936
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Monterey Version 12.7.6 (Build 21H1320)
Total RAM size [MB]: 17,180MB
Resident memory: 456MB
Peak memory usage: 918MB
RAM size reserved for the Undo / Redo stack: 1,074MB
RAM size occupied by the Undo / Redo stack: 0MB
OpenGL installation
GL version: 4.1.0 (4.1 INTEL-18.8.16)
Profile: Core
Vendor: Intel Inc.
Renderer: Intel Iris Pro OpenGL Engine
GLSL version: 4.10.0
Textures compression: Enabled
Mutisampling: Enabled (4 samples)
Installed extensions:
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_texture_mirror_once
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
Eigen vectorization supported: SSE, SSE2, SSE3, SSSE3, SSE4.1

fe60, 2024-09-03:
Seems to be a duplicate of #12937

DerKameramann, 2024-09-03:
> Seems to be a duplicate of #12937

It´s still pending and unresolved. 2.7 was rocksolid even on weak/old Hardware. 2.8 Preview is a pain in tha.

trashmaster9000, 2024-09-03:
@DerKameramann PrusaSlicer 2.7.4 is still avaliable for download.
PrusaSlicer is Open Source, if you want to contribute a fix dont hesitate to do so.

its summer and vacation season, the devs have a life/family, please be patient.

no wonder OSS devs want to quit, suffer from bornout etc. when they have to deal with stuff like your comment.
be patient and refrain from implying the devs dont care about this issue.

DerKameramann, 2024-09-10:
> @DerKameramann PrusaSlicer 2.7.4 is still avaliable for download.
no way ... 

PrusaSlicer is Open Source, if you want to contribute a fix dont hesitate to do so.
there is no need for another codenerd and I dont think you want me to teach how to fix such issues so dont call me up.

> 
> its summer and vacation season, the devs have a life/family, please be patient.
well, a bunch of people, including me, are patient for 3 Months already
i know they work on it - that´s why version 2.8.0 +1 is existent - but the problem persists and some say its parent is a major revision in the developer tool´s world dropping older hardware support. true? probably maybe.
A statement like "nope, old hardware is not supported any longer" or forking a 2.7.x lts version dedicated to older hardware are possible moves.

> 
> no wonder OSS devs want to quit, suffer from bornout etc. when they have to deal with stuff like your comment. be patient and refrain from implying the devs dont care about this issue.
did i say something like they dont care?

19lambert70, 2024-09-10:
slow preview on my System too 
PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (Erzeugungsversion 22000), 64-bit Edition
Total RAM size [MB]: 34,280MB
OpenGL installation
GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2)
Profile:      Core
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
GLSL version: 3.30.0
Textures compression:       Disabled
Mutisampling: Disabled

How to enable the last 2 and how to update GLSL version ? 

Greetings and THX

the-powl, 2024-09-19:
Problem still persistent in Version 2.8.1. Unusable for me unfortunately.

Impact123, 2024-09-28:
> Vendor:       VMware, Inc.

Huh.

FAB1150, 2024-09-28:
> > Vendor:       VMware, Inc.
> 
> Huh.

Not a VM, opengl probably comes with the vmware installation and it stuck as it's the last thing I installed that requires it.

SimonSkapik, 2024-10-02:
Same, the preview gets really slow. 80x80x80 cube with 15% grid infill. 0.2 layers has noticeable fps drop with just 50 layers displayed and after 100 its getting really bad...
[image]

Memory is steady around 400MB for this model and CPU shoots up to about 30% when moving the model.

CPU: Ryzen 5 3600
GPU: GTX 1070
RAM: 32 GB

PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 34,288MB
OpenGL installation
GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2)
Profile:      Core
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 128 bits)
GLSL version: 3.30.0
Textures compression:       Disabled
Mutisampling: Disabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_sRGB
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_shader_integer_mix
GL_EXT_texture_array
GL_EXT_texture_compression_rgtc
GL_EXT_texture_integer
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_ycbcr_texture
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_OES_EGL_image
GL_OES_read_format
</details>

the-powl, 2024-10-06:
From this another problem arises: With PrusaSilcer 2.7 I can't access the new profiles for the new Prusa MK4S. Since I just did the upgrade I'm stuck with the MK4 profiles. This is definitely a thing that needs to be resolved somehow.

DerKameramann, 2024-10-08:
New 2.8.1 made everything worse for me. A simpleformed 10x10x3 cm Skull-Objekt in the G-Codeprieview renders the App almost unresponding.
I´ll give up, going back to 2.7, works like a charm.

dwprod38, 2024-10-18:
Same issue for me since 2.8.0 to 2.8.1 made everything worse for me. A simple formed calibration cube 20x20x20mm (https://www.printables.com/model/591501-calibration-calibration-cube) in the G-Code preview renders the App almost unresponding with 4 to 8 secs to refresh 3D after moving it with cursor... 

What happened since 2.7.4 which works like a charm: I'm wonder to see OpenGL lost its graphic acceleration... ?

In waiting for next 2.8.2 version with optimized source code program, please add missing MK4S in 2.7.4 !!!

Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Sierra Version 10.12.6 (Build 16G2136)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version: 3.3.0 (3.3 NVIDIA-10.2.37 310.90.10.05b54)
Profile: Core
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression: Enabled
Mutisampling: Enabled (4 samples)

Installed extensions:
GL_APPLE_client_storage GL_APPLE_container_object_shareable GL_APPLE_flush_render GL_APPLE_object_purgeable GL_APPLE_rgb_422 GL_APPLE_row_bytes GL_APPLE_texture_range GL_ARB_ES2_compatibility GL_ARB_blend_func_extended GL_ARB_draw_buffers_blend GL_ARB_explicit_attrib_location GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_occlusion_query2 GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_separate_shader_objects GL_ARB_shader_bit_encoding GL_ARB_shading_language_include GL_ARB_texture_cube_map_array GL_ARB_texture_query_lod GL_ARB_texture_rgb10_a2ui GL_ARB_texture_storage GL_ARB_texture_swizzle GL_ARB_timer_query GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ATI_texture_mirror_once GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_depth_bounds_test GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_texture_compression_s3tc GL_EXT_texture_filter_anisotropic GL_EXT_texture_mirror_clamp GL_EXT_texture_sRGB_decode GL_NV_texture_barrier

etmmafl, 2024-11-10:
Same here

```
PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 12,766MB
OpenGL installation
GL version:   4.0.0 (4.0.0 - Build 10.18.10.5059)
Profile:      Compatibility
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 2500
GLSL version: 4.0.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_subroutine
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_map_texture
GL_INTEL_performance_queries
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_debug
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
</details>

```

the-powl, 2024-11-28:
Problem still persistent in Prusaslicer 2.9.0-alpha1
[image]

GPU load goes up to 100% while orbiting the camera. It's still completely unusable for larger models. If this doesn't get resolved I might need to sell my printer because I can't use the new upgraded Prusa MK4S profiles. I also don't want to invest in a 500$ graphics card only because of this bug.

Please please please, get this resolved 🙏🏼

bombela, 2024-12-01:
PrusaSlicer used to have decent performance. Now its often literally unusable because of the lag and stuttering. It is not only the 3D view. Merely scrolling the part list (with like a dozen parts) freezes and stutter as well.

PrusaSlicer
Version:   2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 17,105MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 565.90)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3070/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 329 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

lordjaxom, 2024-12-17:
I can confirm this has gotten worse with PrusaSlicer 2.8.1 on Linux with the amdgpu kernel driver.

On another Linux with an old NVIDIA card and the noveau (open source) driver the performance is the same as it was in the previous versions (very good) though.

dwprod38, 2024-12-20:
Same issue with 2.9.0: when previewing G-code of both parts, important lag 3D moving. But when changed filament of one of two parts without slicing (ghost part). The moving 3D is better... See attached video part.

https://github.com/user-attachments/assets/6283f89b-73e6-4408-9c24-29e0eebe03cf

nmschulte, 2024-12-20:
I tested this with clean builds all the way back to tag [version_2.6.0](https://github.com/prusa3d/PrusaSlicer/tree/version_2.6.0), and the issue was still present.

Also, I find it interesting that when viewing the scene from _below_ the build plate, the performance (FPS) is significantly better.

This is on AMD GPU, under Wayland (Sway; XWayland), Mesa/DRI stack: `AMD Ryzen 7 PRO 5850U with Radeon Graphics`
```
PrusaSlicer
Version:   2.8.1
Build:     PrusaSlicer-2.8.1+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.11.10-amd64 x86_64
Total RAM size [MB]: 16,061MB
OpenGL installation
GL version:   4.6.0 (4.6 (Core Profile) Mesa 24.2.8-1)
Profile:      Core
Vendor:       AMD
Renderer:     AMD Radeon Graphics (radeonsi, renoir, LLVM 19.1.4, DRM 3.59, 6.11.10-amd64)
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
[... 170 lines cut ...]
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_const_bandwidth
GL_MESA_texture_signed_rgba
GL_NVX_gpu_memory_info
GL_NV_alpha_to_coverage_dither_control
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_NV_shader_atomic_int64
GL_NV_texture_barrier
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_S3_s3tc
</details>
```

darkpoetry, 2025-01-12:
2.9.0 - flathub version also unusable.

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0+flathub.org

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.1.0-28-amd64 x86_64
Total RAM size [MB]: 33,156MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 535.183.01)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1050 Ti with Max-Q Design/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 315 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_std430_layout
GL_NV_uniform_buffer_unified_memory
GL_NV_vdpau_interop
GL_NV_vdpau_interop2
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
</details>

djhrum, 2025-01-13:
Same problem.

OS: OpenSuse Leap 15.6
GPU: Intel irix

When it shows all slices lines programm use almost 100% cpu. I can manipulate oblject, but this is not usable.
For example when I press left button and move mouse, nothing happenes for a ten seconds and after that model will be rotated in the random position.

But in the "one layer" mode everething works fine.

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0+flathub.org

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.4.0-150600.23.30-default x86_64
Total RAM size [MB]: 16,509MB
OpenGL installation
GL version:   4.5.0 (4.5 (Core Profile) Mesa 24.3.1 (git-c815d651b8))
Profile:      Core
Vendor:       Mesa
Renderer:     llvmpipe (LLVM 19.1.6, 256 bits)
GLSL version: 4.50.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)

<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
[... 164 lines cut ...]
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_ycbcr_texture
GL_NVX_gpu_memory_info
GL_NV_alpha_to_coverage_dither_control
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_NV_shader_atomic_float
GL_NV_texture_barrier
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
</details>

JKinch03, 2025-03-08:
I believe it has something to do with how the slicer is rendering the lines. I started noticing it around 2.8 like everybody else, which I'm pretty sure is when they switched over to rendering everything in vector format rather than pixels. This would make sense why the layers are causing an issue, because the program wants to render every single travel and movement as a vector.

Here's my specs for comparison:

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 16,737MB
OpenGL installation
GL version:   4.6.0 (4.6.0 - Build 31.0.101.2130)
Profile:      Core
Vendor:       Intel
Renderer:     Intel(R) Iris(R) Plus Graphics
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 171 lines cut ...]
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

bombela, 2025-03-09:
> Installed extensions: GL_3DFX_texture_compression_FXT1

I don't have issues with the 3D rendering. I think I have got a beefy
enough GPU maybe? But it's the GTK UI. Everything is so slow as soon as a
few objects are loaded, it's barely usable. Like scrolling any lists is
almost impossible. Clicking on something lags by half a second. Searching
through settings doesn't work. And piles up a non fonctional auto
completion menu per key stroke that I have to close one by one. With the
horrible lag. The 3D itself is reasonably usable except that for some
reason mouse movements are way too fast. But this last bug has been there
for year. Apparently we don't deserve a mouse sensitivity setting. Changing
tabs in the setting lags. Selecting objects in the 3D view lags. Everything
lafs. It's lag city really.

So in short the UI is so slow, lags so much, it's is really hard to use.
Maybe there is something off on my machine specific to prusa slicer. But it
wasn't as bad before. It just got more lag for every releases. And at this
point it's bad enough that I am looking at other slicers even though I
purchased Prusa.

TomaMihai, 2025-03-18:
Same here. Jumped from 2.7.0 (working fine on my old computer) straight to 2.9.0 (then 2.9.1) and the 3D view is mostly unusable due to lag.

PrusaSlicer
Version:   2.9.1
Build:     PrusaSlicer-2.9.1

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 17,086MB
OpenGL installation
GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2)
Profile:      Core
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
GLSL version: 3.30.0
Textures compression:       Disabled
Mutisampling: Disabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_sRGB
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_shader_integer_mix
GL_EXT_texture_array
GL_EXT_texture_compression_rgtc
GL_EXT_texture_integer
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_ycbcr_texture
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_OES_EGL_image
GL_OES_read_format
</details>

DerKameramann, 2025-04-04:
Working well on my Z840, Nvidia1080 Win10 Workstation.

Unusable on my 8GBram 2010 MBP with Montenery, now absurd unusable on the OC patched Ventura due to more OS Load to the Hardware.
2.7 still working well. Strange. I think Prusa silently dropped the Support to older Grafikcards @2.9 @support_older_hardware

bombela, 2025-04-04:
I have an RTX3070 on win10 and all the UI lags terribly.

On Linux I cannot even run prusaslicer anymore because it now requires flat
pack. But flat pack slows down the shell on my whole system because of a
bug in the completion package coming with it. Furthermore flat pack makes
prusaslicer take forever to start anyways. This is on an AMD ryzen CPU/GPU.

TomaMihai, 2025-05-19:
> Hi, please attach your system info to provide us with more data. It may help us. If you export the G-code and then open it in a G-code viewer, is the issue observed as well? [image]

G-Code viewer moves just as slow. Also on 2.9.2.

Confirmed "ghost part" (change settings while in Preview mode) doesn't have the "slow as molasses" issue, moves great.

PrusaSlicer
Version:   2.9.2
Build:     PrusaSlicer-2.9.2

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 17,086MB
OpenGL installation
GL version:   3.3.0 (3.3 (Core Profile) Mesa 12.0.0-rc2)
Profile:      Core
Vendor:       VMware, Inc.
Renderer:     Gallium 0.4 on llvmpipe (LLVM 3.6, 256 bits)
GLSL version: 3.30.0
Textures compression:       Disabled
Mutisampling: Disabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robustness
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_sRGB
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_shader_integer_mix
GL_EXT_texture_array
GL_EXT_texture_compression_rgtc
GL_EXT_texture_integer
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_KHR_context_flush_control
GL_KHR_debug
GL_MESA_pack_invert
GL_MESA_texture_signed_rgba
GL_MESA_ycbcr_texture
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_OES_EGL_image
GL_OES_read_format
</details>

k1-801, 2025-06-28:
In my research, the performance is mainly degraded by rapid mouse events causing a complete redraw of the UI for *each*.

The easiest solution I found is recording the time of the redraw, and if the time since the last redraw is less than a set timeout, a redraw should NOT happen.

Then, a timer should be added, to fire an additional event after the last redraw, so if there were any events that should have caused a redraw, but didn't because of the timeout, another redraw covers the changes.

Having a timer constantly causing redraws may have it's own performance impact, but it should be less than what we are currently experiencing. If this is a concern on it's own, the timer should only fire once, and then stop, if there were no changes between a redraw caused by a timer event and the previous redraw. If there WERE any changes, another timer event should be placed - so it will redraw until there are no changes and then stop. Additionally, the timer can be stopped early if the window is out of focus.

As a bonus, it should fix the way UI legends contract when preview modes change (the legends adjusts faster when mouse moves, and it may not adjust at all if mouse doesn't move -> no events arrive to cause a redraw).

dakahler, 2026-01-17:
Ok, the problem is with the Mesa OpenGL implementation. If you launch the app with --no-sw-renderer, which forces Mesa off, it runs much better.

Specifically, I often run via RDP, and [this check](https://github.com/prusa3d/PrusaSlicer/commit/88059baddba34fafd9952775491732192999fb3d) added by @bubnikv forces Mesa (aka software rendering), even though the reported OpenGL implementation over RDP is 4+ and works fine. Then, if I leave it running but start using the computer non-remotely, it has already started in software rendering and so is stuck that way.

I propose removing the SM_REMOTESESSION, because if it's being forced to ogl 1.1 the following check will catch that. But I don't have the historical context on why this check was added.

dakahler, 2026-01-18:
https://github.com/prusa3d/PrusaSlicer/pull/15163

closed 2026-09-29 by prusaslicer-bot[bot]