prusa3d/PrusaSlicer #13078: Add option to disable "configuration update" nags

buhman, 2024-07-12:
**Is your feature request related to a problem? Please describe.**
I currently have all update checkboxes in the configuration wizard unchecked:
[image]
However, despite this, I am still given notifications for both 2.8.0, as well as notifications for "configuration bundles":
[image]
Additionally, every time I open the configuration wizard, I am re-nagged with the "configuration bundle" update.

**Describe the solution you'd like**
It should be possible to configure PrusaSlicer to not check for updates of any kind.

It should be possible to configure PrusaSlicer to never offer updates, even when the metadata for that update previously exists in local caches: e.g: if update-checking is at first enabled, then later disabled, update nag-dialogs should never be spawned.

This also includes the lower-right corner notification:
[image]

**Describe how it would work**
The configuration checkboxes for this already exist, they merely appear to have no or partial effect.

**Describe alternatives you've considered**
Clear all PrusaSlicer local storage and apply operating-system level restrictions on PrusaSlicer's outbound network access.

themanyone, 2024-07-13:
Searching around the code for check updates locates this suspicious section

 ```
831 #if 0 // This code was moved to EVT_CONFIG_UPDATER_SYNC_DONE bind - after 
     preset_updater finishes synchronization.
 832         if (! this->check_updates(false))
 833             // Configuration is not compatible and reconfigure was refused by the 
     user. Application is closing.
 834             return;
 835 #endif
 836⬥        CallAfter([this] {
 837             // preset_updater->sync downloads profile updates and than via event 
     checks updates and incompatible presets. We need to run it on startup.
 838             // start before cw so it is canceled by cw if needed?
 839             this->preset_updater->sync(preset_bundle, this,
     std::move(plater()->get_preset_archive_database()->get_selected_archive_repositories
     ()));
```

 Setting a breakpoint with a debugger shows that this part, `CallAfter([this] {` is indeed being called. And the ` if (! this->check_updates(false))` is never processed.  The comment says it (th' check, dude) was moved to EVT_CONFIG_UPDATER_SYNC_DONE. But of course it wasn't.

```
1447         Bind(EVT_CONFIG_UPDATER_SYNC_DONE, [this](const wxCommandEvent& evt) {
1448             this->check_updates(false);
1449         });
```

Aha~~!
 
 Well, that was easy.

themanyone, 2024-07-13:
In `GUI_App::check_updates()`, we see that it is indeed doing `PresetUpdater::UpdateParams::SHOW_NOTIFICATION` even when not invoked by user.

```
3504 bool GUI_App::check_updates(const bool invoked_by_user)
3505 {   
3506      if (invoked_by_user) {
....
3522     }
3523
3524     PresetUpdater::UpdateResult updater_result;
3525     try {
3526         preset_updater->update_index_db();
3527         updater_result = preset_updater->config_update(app_config->orig_version(),
     invoked_by_user ? PresetUpdater::UpdateParams::SHOW_TEXT_BOX :
     PresetUpdater::UpdateParams::SHOW_NOTIFICATION,
     plater()->get_preset_archive_database()->get_selected_archive_repositories());
3528         if (updater_result == PresetUpdater::R_INCOMPAT_EXIT) {
3529             mainframe->Close();
3530             // Applicaiton is closing.
3531             return false;
3532         }
```

themanyone, 2024-07-13:
Added to my ever-growing pull request https://github.com/prusa3d/PrusaSlicer/pull/12994

buhman, 2026-09-04:
This is probably still valid.