prusa3d/PrusaSlicer #13079: PrusaSlicer configuration wizard "Updates" checkboxes have a random state.

buhman, 2024-07-12:
**Step 1:** 
Clear all checkboxes:
[image]
**Step 2:**
Click "Finish". Do not restart PrusaSlicer.
**Step 3:** 
Re-open the Configuration Wizard
**Step 4:**
Click "Don't Install" on the configuration update nag dialog
[image]
**Step 5:** 
Click "Updates"
**Step 6:**
Observe that the checkbox state has changed:
[image]
**Step 7:**
Uncheck all checkboxes
**Step 8:**
Click "Finish". Do not restart PrusaSlicer.
**Step 9:**
Re-open the Configuration Wizard.
**Step 10:**
Click "Don't Install" on the configuration update nag dialog
[image]
**Step 11:**
Click "Updates". Observe that the checkbox state has been re-randomized:
[image]
**Step 12:**
Go back to step 1

**Describe the solution you'd like**
The "Update" checkbox state should not be randomized every time the Configuration Wizard is started.

**Describe alternatives you've considered**
Lobotomize all of this update code out of PrusaSlicer.

buhman, 2026-09-04:
This is probably still valid.

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion https://github.com/prusa3d/PrusaSlicer/discussions/16141.

closed 2026-10-21 by Jan-Soustruznik