prusa3d/PrusaSlicer #13083: Improvements to Filament>Shrinkage Compensation (Expert Mode)

bryn51, 2024-07-13:
**Is your feature request related to a problem? Please describe.**
I use filament Pokemaker Co PA (PA6) which expands when printing and contracts when cooled. I find that (with NO Shrinkage compensation) the printed model HOLES are shrunken, when the gross dimensions of the model are slightly expanded.  This causes fitting issues when the model is installed in an assembly.

**Describe the solution you'd like**
I would like to know how the shrinkage compensation applies to holes better than I currently do. I would like the shrinkage compensation to handle holes better when adjusting for shrinkage. The shrinkage compensation should make holes larger, and the gross dimensions smaller than the original model.  There should be a dimensional test model supplied along with instructions on its use, to allow calculation of the Shrinkage Compensation straightforward. I note this may not be a programming task, but it should be coordinated by associated parties as part of the overall delivery of this facility.

**Describe how it would work**
The shrinkage compensation factor should make holes larger, and the gross dimensions smaller than the original model.

**Describe alternatives you've considered**
Before this facility was available, I applied scale factor (plater tab) to the model, with different values of each of X, Y and Z vectors.
This method does not deal with holes in the correct way.
I also used the modelling tool (Fusion 360) to achieve the same end.  

**Additional context**
These types of adjustment do not allow for the fact that different filaments expand and contact in different amounts.
Adding this facility to Prusa Slicer and stored with the filament profile is a way of making the required adjustment in an easier way, without modifying the original model dimensions, and since the adjustment is made per filament, it better accounts for the different materials behaviour.  The adjustments are stored in the filament profile for use on  other projects.

I grant that it is entirely possible that Prusa Slicer is already doing as requested. But, since its not documented, I do not know that. 
So, a big part of what I am requesting is for better documentation.

Vipcioo, 2024-07-13:
And I'll add that in the shrinkage compensation settings I cannot enter values ​​that reduce the printout (like -0.7%), only just enlarge it.
For me, in Cura it is more transparent and clear, because 100% is the middle of the scale, above means bigger the print, and below means print smaller.
Yes, I can set the engine steps, but I can do it more precisely by scaling the resulting model during calibration. 
Especially in the delta system, which is already largely ignored.

VH-fixer, 2025-04-08:
There is more than shrinkage involved to get a correct size on outside features and inside features. 
Shrinkage needs to be applied to the model definition before slicing calculation.  Supports do not need it.  They just need to adapt to the updated model.

Each filament used with MMU/XL multifilament must ncorporate have shrink factor.    This should come from the filament set of parameters.    Not a user test.   Users should have the option to override the value if they are seeking to hold extreme tolerance values. 

Small holes being too small are also a function of the printers  XY resolution also.  Current .08mm resolution for XY on the MK4/Core One series needs to be factored in also.    PrusaSlicer printer definition needs to be expanded to include the minimum move values. 
PrusaSlicer then needs to be able to factor in how the minimum movements effect or crop the hole size also.   This has been submitted in another enhancement request. 

One more step to a professional slicer.

bryn51, 2025-04-08:
#Ok, firstly the requested enhancement has aleeady been implemented so its moot.

Secondly this enhancement specifically to account for residual dimensional expansion/shrinkage after cooling caused by coeff of exp in the first instance.
This is an effect of the material chosen to print one instance of a model and which may differ for a different material, therefore inappropriate to deal with by changing the model.   
There are other opportunities in print profile to handle the various other effects.

The one remaining effect I'd like to see fixed in PS is the fact it shrinks interior hole walls by ~0.2 mm every time on every model on every material. But no response from PS team on that one.

bryn51, 2026-09-18:
OK i would like this issue to remain open. The core is consistent shrinkage of holes by the slicer. It has been extant for many years and yet no attempt made to correct it. For sure the slicer knows of the holes as it does other things like make solid plastic near a hold where it finds one. So what possible objection could there be to have the slicer tell the printer precisely the correct position of hole walls ? Has this been addressed in PS 3.0 ? We do not know, as nobody has said. If not let this issue remain open until it has expressly been addressed.

VH-fixer, 2026-09-19:
Unless 3.x has addressed this, this is a needed funciton.