prusa3d/PrusaSlicer #13096: Add "View Cube" to manipulate current view perspective

bryn51, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
Its a usability Issue: When manipulating the view of an object with intentin of addng "paint on" supports, its pretty difficult to get the right perspective and zoom in on a point that needs supports added. This is because when the user zooms, the display of the object goes off to the right or the left, and there seems to be no way to get it to zoom straigh tforward towards the user.

**Describe the solution you'd like**
I would like Prusa Slicer to have view manipulation somewhat similar to what we see in Fusion 360 (which has) 
(a) a view cube to orient the display, with a "home" button, to show the model on the bed with the bed in a standard orientation.
(b) a pivot point shown on the display which can be moved around using the mouse.
(c) a "fit" commend button somewhere to centralise the object in the display and zoomed to fit the screen.

**Describe how it would work**
Open a model in Fusion 360.  Manipulate the display using the view cube  and observe how it works.
Make Prusa Slicer work like that, close as you can.

**Describe alternatives you've considered**
I have tried over a couple of years to try to use the display functionality how it already works, and find that the wierd way it currently works is counter-productive. It needs to be made easier and thus more productive.

**Additional context**
Whilst this request pertains expressly to the "paint on supports" functionality, it would be just as useful in other contexts as well.

neophyl, 2024-07-15:
This is a duplicate request. For example https://github.com/prusa3d/PrusaSlicer/issues/9083 or https://github.com/prusa3d/PrusaSlicer/issues/12323  

Some of it is already implemented though.  Your item C for example. Select an object and press Z, which will zoom to that object.  If no object is selected then it will zoom to all objects.  B will also zoom to the bed.
While there is no Home button the numbers 0-6 can be used to go to specific views.

bryn51, 2024-07-15:
> This is a duplicate request. For example #9083 or #12323
>
> Some of it is already implemented though. Your item C for example. Select an object and press Z, which will zoom to that object. If no [...]

Thanks for responding.

Yes, the basic issue is the same in each case.  But the differnet folks have requested different solutions.  The fact there are 3 requests for this kind of functionality must raise the weight of it towards any decision to make changes.

The solutions you suggested are not available to all users, as those commands are not on any menu. You just have to know them.
We do have the "View" menu which does not list them.  The idea of having a UI run by single letter commands is a pretty anachronistic approach, so at the very least, if you are serious to promote the existing functionality as a solution, would be to make those commands visible on a menu (The View Menu), which should be a minor change (at the cost of some string space memory).

neophyl, 2024-07-15:
The fact that there are requests for multiple ways to do things just shows that there is no standard way to do things.  No matter how much people may wish it.  :)  What works for one person might not for another.  You either have to make the entire UI configurable like Blender or people just have to learn to use what is available to them.  

As for the effort involved to do that and if its worth that effort is down to Prusa.  Personally I'd prefer if effotr was spent fixing actual slicing issues but it does seem like people like shiny instead.

I don't know what's available to Mac or Linux users (your issue does not list your operating system or even what version of Prusa Slicer you are currently using) but for Windows the shortcuts are listed in the Help>Keyboard shortcuts listing (which itself has a shortcut of '?').
[image]
[image]

Also they ARE listed under the menu items at least for version 2.7.x on a windows based install.
[image]

bryn51, 2024-07-15:
Aha,  Well I have never before seen "keyboard shortcuts" its well hidden.  Like I said It needs to be on a menu.  For there to be a "short cut" there needs to be first a "cut", and its not there.
As to what is already on the "View" menu, if thats what you think will satisfy this request you do not understand the request.  I am asking because what those various views do: When I select one of them And zoom with the mouse or trackpad, the model zooms off to the right or left, but not straight towards me.  I selected Fusion as an example, but might as well have selected a dozen other Modelling tools, since all have some kid of tools for adjusting the orientation view easily.  But Fusion is what I use, and thousands of others. I grant there are other tools, also with thousands of users.

And If I should wish to rotate the model, good luck with that.  It can be done, but requires lots of jiggering the mouse around. A mother might love the functionality, but its not productive.

This is not about " people like shiny instead" its all about productivity.  As it happens, I see new versions of prusa slicer released all the time, with some new feature I do not want or need. Burgeoning featuritis. But precious little attention to improving usability, fixing the long list of reported bugs, and everything new just tacked on.  Its about time Prusa took a long hard look at those things, as competitors have done.  This is just one thing...

I also wonder why you chose my request to complain about, what about the two prior requests (which I searched for but did not find). Did you also complain about those ?

neophyl, 2024-07-15:
Don't view it as a complaint.  It was not meant to be that.  As you point out there are MANY issues here already and like yourself I think that more effort should be put into fixing things that have basically been around forever.

And yes if you check those issues you will find similar comments about duplicates and cross referencing them.  One of the problem is the number of issues is spiralling out of control and putting links to duplicates makes it easier for the Devs to merge and keep things tidier and easier to manage.  There are a very limited number of people with dev access to the github and a VERY large number of issues.  The result of being a popular piece of software, success breeds its own problems.

Quite a large percentage of issues opened here do not follow the basics requested before opening a new issue.  Many do not bother to search at all, many also do not include all the requested details etc.  Anything we can do to make it easier for the Devs to maintain this place is the goal.  

As for usability vs new features, I happen to agree with you in general.  However functionality of SLICING trumps any and all UI/Usability things on my personal scales.  However neither of us controls the task priorities lol.

bryn51, 2024-07-15:
So, we do have some common ground.

The long and growing list of bugs and features talks not only to the growth in users but also to:
- huge growth this year and last in the number of printers sold, which is what pays for prusa slicer.
- A resource bottleneck. Growth in customer numbers apparently not resourcing the size of the dev team. 
- A change in the flavor of customers. less that are wanting to tinker with their printer and more that just want an easy to use reliable printer, and would like it if the software used to drive it was more intuitive and efficient to use.   

The long list of outstanding dev team work needs to be managed better. 
If someone asks for a new feature then (the wide group) of customers ought to be asked to vote for its priority.  
Github might suit the dev team to use, as it's where the code lives. But it's pretty raw. A better alternative would be something like Jira (Atlassian). And a small group of user friendly folks to run it.
Code like 2.8.0 thats not ready ought not ever be released, but if it is, then the team need to be on the ball to fix issues right away.
There is unfortunately a long list of recent releases of PS (like 2.8.0 ) that seem to have been released when not completed properly. Like 2.5.0 that has the issue with y axis calibration requiring hundreds of users to pull their y axis apart.
So releasing code thats not properly tested in and of itself just causes the list of outstanding work to grow, further exacerbating the issues canvassed earlier. Relying on voluntary testers is NOT the way things ought be done.

wafflecart, 2024-07-15:
> Code like 2.8.0 thats not ready ought not ever be released, but if it is, then the team need to be on the ball to fix issues right away.
There is unfortunately a long list of recent releases of PS (like 2.8.0 ) that seem to have been released when not completed properly. 

On this point a worrying thing I’ve seen just being subscribed to everything on this project and prusa firmware project is you sometimes see a prusa employee ask for the fix to be retested by a user who opened the bug. I’d hope that Prusa is replicating it internally themselves, fixing it and retesting it themselves and not relying on some random people on GitHub to do their QA!

Also the release schedule recently has been too quick?! You see that you’ll get alpha, to release candidates to full release in matter of days. There is no chance that stuff has been tested out in the field in that time frame. The whole purpose of alphas and release candidates is to put it out there and give it time to be installed and run by users for a while. Hence if that’s not done properly and adequate time not given you’ll get a lot of issues back AFTER an official release, as is the case with 2.8.0 and it looks bad..

VH-fixer, 2025-01-17:
Back to the original request.  A view cube along with a visible and setable spin and zoom center point have become industry standards in the 3D viewing industry.   Through decades this has become the quick and accepted way to handle view orientation along with  zoom and spin centering.    
The slicer can do much of this, but as shown though with various request, not easily found and used.  We need a comprehensive,  industry proven method.   
Since much of the code exists, it just needs to be adapted to a view orientation cube with visible and setable center point for spin and zoom.

plutonasa, 2025-03-11:
> Back to the original request. A view cube along with a visible and setable spin and zoom center point have become industry standards in the 3D viewing industry. Through decades this has become the [...]

Orcaslicer has this, and the view cube is such a standard in CAD software, I don't understand how prusaslicer hasn't straight up copied it. It's convenient, intuitive, and not hidden in menus forcing users to memorize shortcuts.

caste1200, 2025-04-18:
can we please get a view cube?

jawaulk, 2025-08-17:
Well, I did a google search this morning to figure out how to do this most basic of tasks when manipulating a 3d model. Figured I turned it off on accident somehow, but google brought me here. I was amazed there was no option to turn a view cube on. I believe it would make it much more intuitive and easier if there was a standard view cube, instead of having to lookup keyboard shortcuts and memorize them.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena