prusa3d/PrusaSlicer #13103: prusaslicer 2.8.0 ironing bug

lemonhole, 2024-07-15:
### Description of the bug

Since version 2.8.0, ironing is applied at an angle to the top infill, that looks bad.

Before version 2.8.0, ironing was parallel to the top infill.

Is this a feature? Maybe should add settings like "ironing angle"?

### Project file & How to reproduce

any print

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Win 11

### Printer model

Neptune 3 Pro

bcegatien, 2024-07-16:
Do you have a picture, a file/project where it is happening ? How to reproduce this bug ?

lemonhole, 2024-07-16:
@bcegatien 

1) add shape, select box

2) look at top surface infill lines' direction (45*)
[image]

3) enable ironing

4) look at top surface ironing lines' direction (* or 90*)
[image]

lemonhole, 2024-07-16:
reproduced this bug in 2.7.1
i almost sure ironing lines were parallel to top infill last time i was using it.
maybe some of the settings cause this behavior

k1-801, 2024-07-31:
> reproduced this bug in 2.7.1 i almost sure ironing lines were parallel to top infill last time i was using it.

Ironing always was along the X axis. Research shows that ironing works best when it is at 45 degree angle to the infill pattern, so the default 45 degree angle of infill, so everything is built the way it's supposed to be.

closed 2026-09-29 by prusaslicer-bot[bot]