prusa3d/PrusaSlicer #13104: Menu system - backwards progress

s-corlett, 2024-07-15:
### Description of the bug

I just updated to the lastest PrusaSlicer 2.8,  Sorry, for the rudness, but the menu system is horrible.  Why do I have to click an extra time or two to get to the menu's I've always used.  How can I get the old menu system back in this version of the slicer?  

### Project file & How to reproduce

[air mattress adapter 3.zip](https://github.com/user-attachments/files/16240964/air.mattress.adapter.3.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8

### Operating system

windows 10 / 64bit

### Printer model

Prusa I3 Mk3S

bcegatien, 2024-07-16:
My intent is not to be rude or to point the obvious if you have tried it. 
Have you tried the : 
Preference/GUI/Use custom size for toolbar icons:/Old regular layout with the top bar

[image]

If this is not a fix, can you give a screenshot of the problematic menu ?

s-corlett, 2024-07-16:
I tried that, but didn't see a change, but I didn't restart.  I'll look tomorrow.Never mind me, I'm old and set in my ways. I used to make gui changes in coding just so folks would see the change, but I got over that 20 years ago.I just do a little printing from time to tome when I need a new part or something. I'm not a big user.SamFrom: bcegatien ***@***.***>Sent: Monday, July 15, 2024 9:04:10 PMTo: prusa3d/PrusaSlicer ***@***.***>Cc: s-corlett ***@***.***>; Author ***@***.***>Subject: Re: [prusa3d/PrusaSlicer] Menu system - backwards progress (Issue #13104)
My intent is not to be rude or to point the obvious if you have tried it.
Have you tried the :
Preference/GUI/Use custom size for toolbar icons:/Old regular layout with the top bar
Screenshot.from.2024-07-15.22-00-04.png (view on web)
If this is not a fix, can you give a screenshot of the problematic menu ?

DoubleStrike, 2024-07-24:
**100% agreed - this breaks operating system user interaction standards, and is NOT progress.**  Turning on the option you stated does nothing to solve the issue that menus are hidden behind an extra click and are now no longer easily keyboard inaccessible.

I would add that hiding the menus behind a toolbar tab/button/widget is a really bad UI decision.  It makes menus inaccessible using standard operating system interactions like Alt-F for the file menu, Alt-E for edit menu, etc.

There's no real benefit to doing this tab bar to replace the menu bar, as literally no one was complaining about the 24 vertical pixels the menu bar took up.  If you want to do this new tab bar thing, just put that below the menu, and don't break operating system UI conventions just for "modern design style points."

I think anyone with vision impairments and using a screen reader or any similar form of UI automation will likely be cursing to high heaven right now.  Let's not go backwards on UI progress.

foorschtbar, 2024-07-29:
Same here, i don't like the extra click for the new Menu. The most options i don't use so often, but new `Menu > File > Import > ...` is a backward step...plase change it back or make it configurable.

closed 2026-09-29 by prusaslicer-bot[bot]