prusa3d/PrusaSlicer #13106: Arc fitting not working in general use (large arcs/rings)

KimmoHop, 2024-07-16:
### Description of the bug

Arc fitting does not have any other settings than on/off, so it works only with limited combinations of radiuses and vertex counts. This shows very well with Tinkercad cylinders and rings, where stock cylinders and rings can have no more than 64 sides. Angle between sides is small, but some hard-coded setting (length or deviation?) disables fitting for large radiuses, **where it would benefit print quality the most**.

In included file, outer ring with 40 mm radius has no attempt to use G2/G3. Inner ring with 10 mm radius seems to be made of G2/G3.

Outer ring facets are more clearly visible than inner ring - even without arc fitting - so IMO fitting is done for lesser improvement.

If more settings were available in UI, allowing users to tune wider selection of curves to be fitted to arcs, the feature would be more generic and useful. There are tricks to make Tinkercad cylinders and rings smoother (or at least have more faces), but slicer should be able to detect arcs without additional tinkering. It is also true that _ordinary_ CAD tools output STEP files without faceting problem, but Tinkercad is simple and fast to use (and arc fitting is not built in the slicer for STEP smoothing).

I might try to do this myself, if I get understanding of which arc welder parameters could be used and some sensible range / usable discrete values for them, and make a PR. If someone knows better, how to improve arc fitting, please go ahead ;)

### Project file & How to reproduce

[arc-fitting-test.zip](https://github.com/user-attachments/files/16257108/arc-fitting-test.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10

### Printer model

custom Voxelab Aquila (Ender 3 V2 clone)

KimmoHop, 2024-07-17:
I'm not sure who bound arc fitting tolerance or resolution (the code is not clear, which it **is**) to G-code resolution, but that might be terrible choice.

After cranking G-code resolution up to 0.1, arc fitting starts to work, though Cura Arc Welder plugin does better job. Arc fitting combines G1 and G3 codes where Cura plugin does with fewer G2/G3 codes and almost no G1 between.

How do these map, exactly?
[image]
[image]

Tupson444, 2026-09-19:
still relevant