prusa3d/PrusaSlicer #13118: Multi-color painting enhancements

mikeosh, 2024-07-18:
**Is your feature request related to a problem? Please describe.**
I have a new XL-5T.. Multi-color painting is frustrating.  I'll trying smartfill/brush to change a color, but some small polygons get overlooked and remain the original color.  This is due to smartfill angles or to ploygons that aren't visible when doing the smartfill or brushing.

I end up having to eaxmaine everything at a high zoom factor and use the clipping factor to see everything.

**Describe the solution you'd like**
1) Have a mode to show all color transitions and where they happen.  That way I can find unwanted transitions.
2) Show everything that will be printed by a selected toolhead.  That way I can also find unwanted transistions.

**Describe how it would work**
1) Add a view highlighting all tool change transitions in the print.  Alternatively, show all transitions in a sixth color or blinking.
2) Everything in a single toolhead view - show every polygon in that toolhead color.

**Describe alternatives you've considered**
Can't find anything - I just have to do a close examination at high zoom factors.

**Additional context**
Add any other context or paint drawing about the feature request here.

Jan-Soustruznik, 2024-09-16:
Hello, @mikeosh , could you share some 3mf(project file) with painted model, where is your problem as example. Thank you, have a nice day.

Mikyner, 2024-10-26:
[image]
I need this in Prusa Slicer

closed 2026-09-29 by prusaslicer-bot[bot]