prusa3d/PrusaSlicer #13121: Bridges with holes in them always fail, an idea for how they could be improved

factoidforrest, 2024-07-18:
**Is your feature request related to a problem? Please describe.**
When printing a bridge with a hole in it, which is a fairly common situation when printing bolt/screw holes, the holes cause the bridge to fail. The slicer simply prints the holes first, in midair, and they don't work, and it makes a big mess. Heres an example slice to give an idea:
[image]

**Describe the solution you'd like**
However, it seems perfectly possible to print these structures in midair, with the right pattern.

**Describe how it would work**
If the slicer first printed the bridges it _could_ print that avoided the hole, it would have a much better structure to print the holes second.
Forgive the MS-paint:
[image]
In this *cough* drawing, imagine that the slicer first printed the red sections crossing the gap, then orange, then maybe green, and then finall printed the hole outline, it would end up with a much better result in my opinion. This approach is a bit like a spider-web pattern. 

I lack even the first bit of math skills to implement this, but I can see that it's possible. Thanks!

me25542, 2024-07-18:
This would be nice to have, it would be like stepped geometry holes but better.

factoidforrest, 2024-07-19:
@me25542  interesting technique that I wasn't aware of

https://shop3d.ca/blogs/news/3d-printing-tip-1-designing-better-3d-printed-holes?_pos=1&amp;_sid=d69277192&amp;_ss=r

That's a nice trick to essentially force the slicer to do what I'm describing, but still suboptimal when I think the slicer could do almost as well on its own with the right algorithm.  Nice to know about for when I really need it, though. 

As an alternative, apparently some folks also just include a one layer thick area without the hole and then print the hole on top and cut out the layer with a knife after. 

These solutions serve well to demonstrate that it's a common issue.

iviar, 2025-05-20:
+1 after I just published https://www.printables.com/model/1301743-example-for-improvement-suggestion-to-slicer-bridg
The filling image below is edited:
[image]
the current PrusaSlicer 2.9.2 produces instead:
[image]

iviar, 2026-09-23:
I can't test version 3 at present to see the 3 slicing behavior for this case, but I don't consider appropriate to close by general automation this essential improvement suggestions which should be addressed specifically

closed 2026-10-06 by Jan-Soustruznik

factoidforrest, 2026-10-07:
I believe this issue was closed by mistake, @Jan-Soustruznik . That suggestion is stating a similar problem, but with a far worse solution of simply printing a skin under the hole. If anything, that issue should be closed in favor of this one. 

Wave overhangs would handle this perfectly and they show that clever slicing can totally change printing capabilities and revolutionize or eliminate these problems entirely. I believe this issue should stay open until the slicer actually has a functional solution. Thank you.