prusa3d/PrusaSlicer #13123: text shape with korean font

mnstn, 2024-07-19:
### Description of the bug

When I added text shape on a stl, there was an error about setting korean font and input korean.
After I selected korean forcedly, The sliced file was good.
The problem is I cannot open the 3mf file again.
[image]

### Project file & How to reproduce

[top_plate_20181108-61-13m4lkw.zip](https://github.com/user-attachments/files/16313199/top_plate_20181108-61-13m4lkw.zip)
1. set english font
2. type korean to text shape
3. change it to korean font
4. save 3mf file and open again
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8-win64

### Operating system

window11 pro

### Printer model

prusa mk4

Jony01, 2024-08-13:
Thanks for report and Sorry.

You are right using this font creates 3mf in bad shape, which is unopenable for PrusaSlicer any more.

It is connected with encoding of wxFont descriptor, which I baddly expected that will contain only english character.
In your case it contain some not utf8 character.

Fast fix: In file top_plate_20181108-61-13m4lkw.3mf\Metadata\Slic3r_PE_model.config
attribute `font_descriptor` can't contain bad character
[image]

Manualy fixed file to be able open it:
[top_plate_20181108-61-13m4lkw.zip](https://github.com/user-attachments/files/16599365/top_plate_20181108-61-13m4lkw.zip)

Jony01, 2024-08-14:
@mnstn could you please provide us your 'korean' font file for test

Jony01, 2024-08-14:
I create fix. 
build for windows version to test it. [[link](https://slicerbuilds.prusa3d.com/fs/PrusaSlicer-2.8.0+127-fs-SPE-2442-win64-g8a2f19060-202408140751.zip)]
Plese Confirm that in this version is possible to store text written in 'korean' fonts and re-open the new created 3mf file.
I am sorry, I can't add option to open of previous created 3mf. (it must be manually fixed)

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena