prusa3d/PrusaSlicer #13125: "Negative volume" adds positive volume in 2.8

Stooovie, 2024-07-20:
### Description of the bug

I select the part, press "Add negative volume/SVG", select a SVG, that imports properly, but volume is still being added to the original object, not substracted from it.

[Video](https://www.youtube.com/watch?v=mMJEaFLTzNE)

### Project file & How to reproduce

1. select part
2. Add Negative volume/SVG

SVG is positive, not negative

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8

### Operating system

MacOS 14.5

### Printer model

Ender 3

u89djt, 2024-07-20:
Fellow user here agreeing that that's a bug.  It looks like the slicer ducks out of the addition of the object and the negative volume classification and just adds the svg as an object to the bed when you click on the object on the bed rather than the object in the list.
While you're waiting for the bug to be asssigned and resolved, you can work around it by right clicking the disc in the objects list rather than on the bed to add the negative volume. If you going from the start point of attempting to add it by right clicking the object on the bed and ending up with a wine glass object per se, then you need to merge the disc and the svg glass and change the type of the glass to negative volume. I've done that here. You'll see that the negative object appears indented alongside the original disc. Your screenshot is cropped, but I can see that the disc and the glass are not indented, and so appear as their own separate object. Perhaps negative volumes should be specifically marked.
[image]

kubispe1, 2024-07-31:
This is a issue related to: #12915
Should be fixed with SPE-2386

lukasmatena, 2024-09-12:
Duplicate of #12915. Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena