prusa3d/PrusaSlicer #13127: version 2.8.0 spans a lot of msedgewebview2 threads - high cpu usage when idle

jiri-jirus, 2024-07-20:
### Description of the bug

version 2.8.0 spans a lot of msedgewebview2 threads this results in unnecessary high cpu usage when idle. I have set up physical printer with Octoprint API key

### Project file & How to reproduce

standard Prusa MK3 printer with Octoprint over LAN with API key - physical printer. Task manager show a number os msedgewebview2.exe processes which consume unnecessary high percent of cpu.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 10

### Printer model

Prusa MK3

MFHFozzy, 2024-08-02:
I came here to report the same.
Theres a cause, and easy fix.
Slic3r sued to default to the 3d editor view. That view does not spike the CPU or GPU.
Now it defaults to the layer preview mode, and *that* mode is what spikes the CPU/GPU/msedgewebview2  threads.
Switching back to editor view immediately fixes the problem.
Make Slic3r default back to editor view.

jiri-jirus, 2024-08-03:
> I came here to report the same. Theres a cause, and easy fix. Slic3r sued to default to the 3d editor view. That view does not spike the CPU or GPU. Now it defaults to the layer preview mode, and [...]

sorry, but that is not a fix, but a workaround. msedgewebview2 can never produce spikes & excessive cpu usage. if it is doing this stuff, is it a bug!

mrneutron42, 2024-08-05:
I'm experiencing the same with PrusaSlicer 2.8.0 in Windows 10.  
I found that CPU usage is high when in the Printer Files tab.
If I switch back to the Dashboard tab, the usage goes down again.
[image]

fe60, 2024-08-16:
According to '@buhman's [comment on #13236](https://github.com/prusa3d/PrusaSlicer/issues/13236#issuecomment-2282714473) it's related to #12968

lukasmatena, 2024-11-21:
Improved in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena