prusa3d/PrusaSlicer #13128: printing of the draft tower

Wlad123w, 2024-07-20:
### Description of the bug

There is an error when printing the draft tower!! Invalid speed in ' G1 2147483648 version 2.8 klipper

### Project file & How to reproduce

[banka v1.zip](https://github.com/user-attachments/files/16321552/banka.v1.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2,8,0

### Operating system

windows 10

### Printer model

gost6

murk-sy, 2024-07-22:
From what I can tell the error is in your toolchange gcode, which is not a slicer issue. Can't be 100% sure since you didn't provide a project file.

[image]

ojrik, 2024-07-22:
Greetings. I was able to replicate your project. The error is in your settings, namely that you have not filled out the parameters for cleaning the extruder. as you can see in my photos. I will attach an experimental cleaning parameter for the PRUSA MINI, but I have not tested it.
[image]
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]