prusa3d/PrusaSlicer #13129: Visible line on exterior perimeter when generating top surfaces (not sure if similar to benchy hull line)

3dPrinting1127, 2024-07-20:
### Description of the bug

Any ideas on what's causing this issue and how to mitigate it?

[image]
[image]

The line appears where the top interior portion of the model is printed

[image]

The line is most potent on the sides I've shown in the first two images, which also coincide with where the bridge infill is overlapped with the perimeter as shown in the black circles (my infill/perimeter overlap is set to 15%). 

[image]

Here's what the solid and top infill looks like.  I just wanted to emphasize here that these infills are not connected to the exterior perimeter where the line appears, I'm not sure if this means I could rule this out as the culprit though.

[image]

### Project file & How to reproduce

[test.zip](https://github.com/user-attachments/files/16321721/test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusaslicer 2.8

### Operating system

Windows 11

### Printer model

MK3S+

u89djt, 2024-07-20:
(Intrigued fellow user) The regions above and below the line are different textures, so could it be a speed/temperature thing? The upper layers take much longer, so the filament of the outer walls has cooled a lot further before the next extrusion. That reasoning doesn't explain the dark line between them, though.  Unless the mass of the bridging extrusions adhering to it conducts sufficient heat to change the texture. Though as you say, they're not continuous around the perimeter. Hmm.
[image]
You could remove the bridging perimeters from the equation by splitting the walls from the internal main floor by a fraction of a millimetre so the internal volume of the wall isn't interrupted.

u89djt, 2024-07-20:
Here's a version with continuous walls top to bottom to try if you like. I fumbled the connector - you'll have to add that. It doesn't change the difference in layer times, but the bridging isn't involved in that surface now.
[modified test model.zip](https://github.com/user-attachments/files/16322084/modified.test.model.zip)
[image]

u89djt, 2024-07-20:
Here's the mimimum layer time setting - if you can bear the wait, maybe slowing the layers below the internal floor could equalize the surface textures somewhat. 
[image]

3dPrinting1127, 2024-07-20:
>You could remove the bridging perimeters from the equation by splitting the walls from the internal main floor by a fraction of a millimetre so the internal volume of the wall isn't interrupted.

Yeah I'm gonna try this now only for the bridge layer and see if this has an impact.  I'll report back tomorrow and see what the results are for this.  Thanks for the suggestions

3dPrinting1127, 2024-07-22:
Well, it produced the same exact results.  I setup those perimeters for the bridge layer and it appears I was wrong.  Bridging isn't what's creating that line...hmmmm it appears this problem is the same as the hull line then, and I'll just have to live with it

u89djt, 2024-07-22:
Do me a favour and post the 3mf file with your modified version? I'm going to have a look at the benchy hull line. I've only seen it a couple of times across multiple prints of it myself.

u89djt, 2024-07-22:
Suddenly from 51s to 21mins
[image]
I bet it takes a lot less that 21 mins for an extrusion to reach an equilibrium temperature. I'm going to keep chasing this for my own satisfaction. If you're happy calling it a day, I won't flood you with more messages.

3dPrinting1127, 2024-07-22:
Yeah from what the article says it appears that is what's occurring.  I didn't use the new model you made because I made this one 
[test model2.zip](https://github.com/user-attachments/files/16336672/test.model2.zip) right away.  Again, this had zero impact on the line unfortunately.  Also, I do appreciate the help.  I actually sell my printed models so the exterior quality is of utmost importance to me hahah.  Thanks again for the help.

u89djt, 2024-07-22:
No worries, it's handy to learn from someone else's travails :) I wondered about trying to print it with a snug print shield to maybe act as insulation to so things stay at some kind of temperature. I see it's the rear end of a ship - presumably why you've added a connector hole: 
[image]
For other folk who take an interest, playing around with this object is fine so long as it's attributed: https://cults3d.com/en/3d-model/tool/fixed-imperial-nebulon-b-cruiser under a CC BY license
I think we can conclude that it's not a slicer issue, so this probably needs to be closed. I'll post a fix to printables if I come up with one and let you know.

closed 2026-09-29 by prusaslicer-bot[bot]