prusa3d/PrusaSlicer #13131: Add global overrides for Prusa defaults

mikeosh, 2024-07-20:
**Is your feature request related to a problem? Please describe.**
I'd like to use different global defaults - I have to change these in every project file I use

**Describe the solution you'd like**
Overrides for global defaults.

**Describe how it would work**
Add global overrides.  Final values would look like this:

Prusa defaults -> global overrides -> filament values -> project values

Example:
1) I'd like to change infill default from grid to gyroid
Prusa: infill grid -> override: gyroid -> project file

2) print output : add weight, using suffix below as my default
{input_filename_base}_0.4n_{layer_height}mm_{printing_filament_types}_XLIS_{print_time}**_{total_weight}g**.gcode

Note: override fields would be blank unless new override value added.

In settings, perhaps have columns: Prusa default value, override value, filament value, project override value.   Make it obvious whenver a value has changed from the Prusa default.

**Describe alternatives you've considered**
Can't find any other way to do this.

**Additional context**
Add any other context or paint drawing about the feature request here.

PaparazziN, 2024-07-21:
Hi @mikeosh 
I think the best solution to this in your case would be to utilize a 'User Preset'. You can create a preset by simply changing whatever you property you like in 'Print Settings' (you can also create Filament and Printer presets btw) and simply clicking the save/floppy drive icon next to profile name to create a 'User Preset'. These presets work in the way you describe, if I understood you correctly.

rmpratt1, 2024-07-26:
I tend to agree with @mikeosh's idea (for at least some overrides). The infill pattern is a perfect example, I'd prefer my default to be Gyroid and change it for cases I don't want that. Making four "User Presets" to make some changes like that is more likely for mistakes of me hitting the Prusa default vs mine, or when I import a 3MF.

rmpratt1, 2024-07-27:
Hi @PaparazziN, I wouldn't consider this issue closed as your suggestions are merely workarounds rather than an actual solution or a wont fix/implement. 

Regarding your comment about updating the PrusaResearch.ini that does not seem optimal. As you mentioned when a configuration update comes through it would override that file and reset things. 

One implementation I could see is having an additional settings type in the drop down. Using print settings drop down as an example:
```
------ Global user presets ------ 
Global user presets
------ System presets ------ 
0.15mm SPEED
0.15mm STRUCTURAL
0.20mm SPEED
0.20mm STRUCTURAL
------ User presets ------ 
0.20mm STRUCTURAL GYROID
```
The advantage of this setup is that the setting is persisted across application and configuration updates (like user presets) but would also include changes the Prusa team makes to configurations which I have not updated. If I use user presets as you suggest and the Prusa team makes changes in the configuration profiles I have to go into each user preset and revert or update those changes. Beyond just the infill type I have spent time calibrating my printer across several dimensions [Ellis' Print Tuning Guide](https://ellis3dp.com/Print-Tuning-Guide/), which I would like applied consistently.

VibroAxe, 2024-12-13:
I support the need for this global overrides, creating user presets is clunky for the generic case and loses the automatic upgrade of presets from prusa. It also increases the overhead, my example to add

1) Ability to switch default support mode to Organic - I want this to happen on everything and not always have to hit transfer

neophyl, 2024-12-13:
> I support the need for this global overrides, creating user presets is clunky for the generic case and loses the automatic upgrade of presets from prusa. It also increases the overhead, my example to add
> 
Not sure you follow how the presets currently work.  If you create a User preset that is based on one of the Prusa defaults, say you changed your infill type to something else like cubic as an example, then saved it.  You would get the new infill when you select that preset obviously.  However if Prusa later changed the profile that your User preset is based on to change say a speed setting then your User preset would ALSO pick up that change.  UNLESS you have chosen to use the Detach from System preset button and then saved it.  That button option breaks the inheritance link.  

Basically your user one contains the link to the original profile and any specific changes you have made.  So if Prusa also later updated the default infill then your preset wouldn't use it as it would contain the setting for infill, in this example Cubic as infill.  That's how all the profile inheritance works.  

The previous suggestion of editing the PrusaResearch.ini is the problematic as that file WOULD be replaced on any update so anything you manually edited in there would be reset.

Just start using User presets properly.  If you want Prusa changes to trickle down then leave the Preset inheritance intact.  If you do not want ANY further Prusa changes to impact your carefully tuned profiles (which is my use case personally) then click that oh so very useful detach button and save.  nd never worry about changes pushed out effecting things again.

castaway, 2025-05-03:
This also irritates me no end - I open the slicer, make the changes I like.. then never save anything (just Transfer every time I switch settigs).. keep the slicer open forever, and then curse a lot should my computer reboot. Certain settings could really do with a global default that applies to ALL print settings, printers, MMU colours etc. Also means I can't save plates/3mf files etc (because no, I don't want to save/lose those modified settings)

castaway, 2026-09-19:
this issue still needs solving (unless its in 3.x already and i missed it)