prusa3d/PrusaSlicer #13137: Prusa Connect logs me out periodically

abatkin, 2024-07-22:
### Description of the bug

Every once in a while (after a couple weeks maybe?), Prusa Slicer reports that I am no longer logged in to Prusa Connect and I need to log in again. Nothing about my Prusa account has changed, so there is no reason why it should ever log me out. This means that I can't print (without shuffling USB sticks) until I have looked up my password and MFA token, which is incredibly frustrated.

### Project file & How to reproduce

Use Prusa Slicer for a while, then one day notice that it's not logged in anymore.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+linux-x64-GTK3

### Operating system

Linux/Fedora 40

### Printer model

MK4

rmpratt1, 2024-07-27:
@abatkin Are you being fully logged out in PrusaSlicer or is your Prusa Connect tab just not displaying anything? I ask because I have not experienced being logged out (Mac OS 14.5) but I experience that the Prusa Connect tab does not display anything other than the Prusa Connect logo.

abatkin, 2024-07-28:
Fully logged out. I think the last time it happened was during a blackout when my internet dropped, but my computer stayed on. But it's definitely happened at least one other time.

arekm, 2024-07-29:
I also experience constant logging out (like after few days I'm logged out; I log in; few days; logged out). MacOS 14.5.

It's **so bad** experience. 

* Prusa Slicer uses integrated browser which doesn't remember my username
* it doesn't have my password manager access (like regular browser), so I can't just allow it to use my password for connect. I have to manually copy password from password manager and enter in PrusaSlicer
* then enter second factor code from phone

again and again every few days.

abatkin, 2024-07-29:
@arekm is right, and that's also a security concern. Users shouldn't be asked to enter credentials with the app, it should be opening the system browser and allowing authentication through there.

Not only does this solve the issue with credentials management (like being able to use a password manager), but it lets the user validate that the credentials are going to the right place. When an application prompts for credentials on its own, there's really no way to know that it is really the application itself and not a spurious popup from somewhere else, but browsers have the URL bar to let people know where they are (plus password managers know which credentials go to which sites, so it won't suggest the Prusa credentials unless it is really the Prusa site).

A great example of this is the Jetbrains developer tools. When it needs to authenticate you, a browser pops open, you log in (or, since it's your usual browser, there's a good chance you are already logged in), it asks you to confirm, and you are done.

That doesn't solve the (original) issue of being logged out of the app (which should never happen), but it would make the whole process less painful.

arekm, 2024-07-29:
Login process was in external browser before but Prusa Team decided to change that in 2.8.0-alpha5 - from changelog:

"The login into Prusa Account is no longer done in external browser window. Instead, the dialog shows directly in PrusaSlicer to streamline the workflow, and the user is thus not forced to switch between PrusaSlicer and the browser."

Why they decided do decrease security and cause such problems? - no idea. That "streamline the workflow" clearly become worse.

mamenama, 2024-08-07:
Another side effect of this is that Google accounts that are Family Link Managed can no longer log in via the app, since they are restricted to logging in via the browser only.

abatkin, 2024-08-11:
Aaand it just happened again (some time over the last ~2 hours). No power/network errors this time. PrusaSlicer was running, I had everything all ready to print. Went and did something else for a couple hours, and when I switched back to the running Prusaslicer instance, it said it could no longer connect.

[image]

Jan-Soustruznik, 2024-08-19:
Similar issue with log off from prusa connect itself. Re-connect token isnt created correctly. Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12951

abatkin, 2024-08-19:
I don't think this is a  duplicate of #12951 - my Prusa Connect tab does NOT go black. Please re-open this issue.

quintesse, 2024-09-15:
Same here, the issue is not the tab going black. In fact the entire Prusa Connect tab keeps working, I just can't do anything that requires authentication. If I, for example, try to delete a file I get "Invalid or missing security token".

This fix here should _at least_ be a way to force PrusaSlicer to log in (at did so at startup so why couldn't it do it again automatically?) Right now I'm forced to exit and restart

m10d, 2024-10-24:
This reproduces frequently for me as well.  Also on linux, but running `2.8.1` release via `flatpak`.  

I have no idea is the new issue @quintesse submitted is the same, I asked them for more details; but note OP (and myself) on linux, quintesse is windows - could be the same.  Quintesse's issue sounds exactly like the oath2 access token (which is always short lived) not being refreshed by Prusa's app using the (long lived) refresh token stored after a password login.

PlumpCat19, 2024-12-06:
This happens to me multiple times a day now. I am using Ubuntu 24.04 and I am starting to consider making my password something very easy to remember so I don't have to find my password in password manager, copy it over and then verify using an authentication app every single day, sometimes multiple times a day.

scmanjarrez, 2025-01-01:
This issue happens to me constantly, and it's very frustrating. I'm running the latest version (2.9.0) on Windows 11 24H2.

Stumpy-finger, 2025-01-09:
I am now having similar issues with my laptop, but can login just fine on my phone

SammyRamone, 2025-01-17:
I also have the same issue of getting logged out very often, sometime on the same day. I am using Ubuntu 22.04 and Flatpak.

robnee, 2025-01-27:
I also have this issue.  in my case it seems related to logging in on multiple devices.  If I start a print from my desktop it seems I frequently (though not always) need to log in to the iPhone app to monitor the print.  Then when I return to my PC after the print finishes I'm logged out on the desktop again.  I have 2 factor enabled and this is terribly frustrating coupled with PrusaSlicer's lack of password manager integration.

I'm not saying add password manger integration.  Rather, please, never log the user out for any reason, ever.  At the very least, if the user's session ends for any reason it would be very helpful to indicate to the user **why the session was ended** in the notification.  If there is any diagnostic info from the failed authentication that could be shared with the user it would help so they could report it when seeking support.

Thay said, my hunch is that this is a problem with the back-end Prusa account handling.  However, I don't know where else to raise issues about the Prusa Connect authentication system except for here.

abatkin, 2025-01-27:
@Jan-Soustruznik Please re-open this

I think there are a number of things that should ALL be done to make this better:
1. Figure out the source of errors that cause PrusaSlicer to _think_ that it's no longer logged in, and fix that
2. PrusaSlicer should ignore certain types of authentication errors, since presumably they are intermittent (we know this for a fact: Sometimes if I have multiple instances on one PC, one will report an error but the others one, and if I restart all of them, I'm still successfully logged in)
3. **Use a real web browser for login** instead of the integrated login. Not only is that more secure, but it means that anyone who has a password manager can log in with a lot less friction

Quyzi, 2026-01-21:
I am having this same issue.  Every time I launch prusaslicer it says Failed to connect to Prusa Account. And I have to login again.  I use a password manager and my password is long, so I have to open another browser and find the prusa account to copy the password.  Either that or I have to walk over to the machine and grab the usb stick.  Both of these are very inconvenient. 

@Jan-Soustruznik This is not a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12951.  This is a completely different issue.