prusa3d/PrusaSlicer #13139: Normal Mode : Print Settings : Elephants Foot Compensation doesnt work

ukwebb, 2024-07-22:
### Description of the bug

With the Normal Mode selected
Elephants foot compensation doesnt work as expected  trying to enter a negative value either is blocked from being typed ( normal keyboard) or if pressed on the number pad which does type, produces an error "Input value is out of range"

### Project file & How to reproduce

using 2.8.0 release on Windows 11 build 23H2
Load any model
Change the working mode to NORMAL MODE
go to PRINT SETTINGS
under SLICING section  
click into text box for ELEPHANT FOOT COMPENSATION
ISSUE 1: try to type the minus symbol using the key next to 0 on the main keyboard - nothing happens
ISSUE 2: type a minus symbol using the key on the number pad and it type the symbol
enter -0.1 - or -1  or -10  and the message INPUT VALUE IS OUT OF RANGE is displayed

Swap to EXPERT MODE
same setting:
ISSUE 3 : minus key on main keyboard is still ignored
but works and value is accepted if using the numberpad minus key

normal mode : [image]

expert mode : [image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0 +WIN64 release

### Operating system

Windows 11 23H2

### Printer model

custom built Duet3d based machine

neophyl, 2024-07-23:
The ability to enter a minus number is a separate issue reported many times.  The current open one with the collection of the duplicates is https://github.com/prusa3d/PrusaSlicer/issues/13043

Besides that though Prusa Slicer has never allowed the use of a negative value.  It has always been a positive value.  The larger the number the more it moves the perimeters in.  You may be confusing this with other slicers such as Super Slicer in which negative moves the perimeters inwards and positive values move the perimeters outwards (like happens with the XY size compensation. This is one of the gotchas you need to be aware of when trying profiles between the 2 slicers.  

Personally to me this has never made any sense.  As logically it should work like the XY compensation but just apply to the first layer.  However that's the way it is in PS currently.  So not a bug but 'by design'.  The problem Prusa would have is that all the existing profiles and saved 3mf files etc would all be 'wrong' if they changed that setting to match how other slicers do this now.

lukasmatena, 2024-08-16:
Thanks to @neophyl for providing the answer. Closing.

closed 2024-08-16 by lukasmatena