prusa3d/PrusaSlicer #13140: Negative Translate position no longer possible in 2.8.0

ecm-bitflipper, 2024-07-23:
### Description of the bug

I used to be able to adjust positions left/right, forward/backward, up/down using either positive or negative translation values.  As of 2.8.0 or thereabouts, the Translate fields don't allow the minus sign.

### Project file & How to reproduce

Load any object.  Select it.  Choose Object coordinates in the right panel.  Place cursor in any of X, Y or Z fields for Translate.  Try to insert a minus sign.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 11

### Printer model

Anycubic Kobra 2

neophyl, 2024-07-23:
Duplicate issue, reported many times.  Current open issue with list of duplicates is https://github.com/prusa3d/PrusaSlicer/issues/13043

lukasmatena, 2024-08-16:
Closing as a duplicate of #13043.

closed 2024-08-16 by lukasmatena