prusa3d/PrusaSlicer #13143: Custom G-Code setting of first layer temperature ignored if other temperature previously set in custom g-code

gonzonia, 2024-07-23:
### Description of the bug

I'm trying to set the custom g-code to heat to 160, mesh bed level, move to home, and then finish heating to the correct first layer temp.  With the custom g-code below however, the first_layer_temperature commands are not exported into the g-code. 

Here is my custom g-code. 
`M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.13.3 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode

M104 S160; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S160 ; wait for extruder temp

G28 W ; home all without mesh bed level
G80 X{first_layer_print_min[0]} Y{first_layer_print_min[1]} W{(first_layer_print_max[0]) - (first_layer_print_min[0])} H{(first_layer_print_max[1]) - (first_layer_print_min[1])} ; mesh bed levelling
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
G1 Z0.3 F720
G1 Y-3 F1000 ; go outside print area

M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp

G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E9 F1000 ; intro line
{else}
G1 Z0.2 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
{endif}
G92 E0
M221 S{if layer_height<0.075}100{else}95{endif}

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}`
 

### Project file & How to reproduce

I see this on all projects. I've attached a project I was using and the the gcode exported that excludes the first layer temperature command. 

[Example.gcode.zip](https://github.com/user-attachments/files/16349553/Example.gcode.zip)
[Custom_Spinner_11_Gyro4_03_SingleColour.3mf.zip](https://github.com/user-attachments/files/16349554/Custom_Spinner_11_Gyro4_03_SingleColour.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+MacOS-arm64

### Operating system

MacOS 14.5

### Printer model

MKS3S with Revo Hotend

plampix, 2024-07-23:
That's because the first_layer_temperature M104/M109 commands are only inserted when filament is set to Prusament PA11.

neophyl, 2024-07-23:
You are missing the point previously raised.  Theres a conditional **if** filament setting in your posted start black that will only add it if the filament is set to Prusament PA11.  Remove that conditional and the endif and it will add those lines in all the time

plampix, 2024-07-23:
Of course, because the M104/M109 with first_layer_temperature are inside the 

{if filament_settings_id[initial_tool]=~/.Prusament PA11./}
 M104/M109
{else}
{endif}

So they don't get executed when something else than PA11 is selected.

gonzonia, 2024-07-23:
Oh! I see now. Thank you!