prusa3d/PrusaSlicer #13144: Internal Perimeters First default sometimes starts on the external perimeter (hole perimeter)

eVenom, 2024-07-23:
### Description of the bug

As you can see in the picture in some cases the external perimeter prints first making overhangs dificult to print.  It seems to be on the external perimeter of a hole.
[image]

it seems to be the case when the external perimeter in question is on the center of the object(hole).

Checking the "External perimeters first" option does not change the behavior on the inside perimeter(hole) and it only changes on the very outside of the objext
[image]

once the perimeter is not part of a hole the the function works as expected
[image]

### Project file & How to reproduce

[Fidget_spiral_cone.zip](https://github.com/user-attachments/files/16353437/Fidget_spiral_cone.zip)

Go to second layer and see the order of the layers, Check and uncheck the "External perimeters first' and see that it doesnt do the internal perimeters first no matter what. 

go the some of the top layers to see that the behavior gets resolved once its not in the perimeter of a hole

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 10

### Printer model

Ender 3

u89djt, 2024-07-23:
(Fellow user having a quick look) I set the scale to 200% and note that the outer perimeter is printed first when the internal perimeters of the outer and inner surfaces of the object merge or disappear.
At 200% scale, at layer 14 the inner perimeters are separate and the perimeters are laid in the order you want. At layer 15, the inner perimeters merge, and the outer perimeter of the inner face is laid first.
The classic perimeter generator (as opposed to arachne) lays them in the order you want over the whole object.
[image]

u89djt, 2024-07-23:
Additional detail: how ever many perimeters you set (try 6), the order is wrong when there's one merged perimeter or no perimeter between the external perimeters at the nearest points between the inner and outer surfaces before the helices separate. That again is at 200% up to the 21st layer. Could be a nice example for someone to look into when working on the arachne generator. 
[image]
Classic always works as intended.

eVenom, 2024-07-27:
I also just tried this in OrcaSlicer and it gets handled as expected in both classic and arachne.

So this is obviously a bug in PrusaSlicer using arachne perimeter generator

u89djt, 2024-07-27:
How do the extrusions look in comparison to in prusaslicer? Screenshots of the same layers created by both slicers with all the same settings - I only just saw that you've picked huge values for width and height - might hold clues.

u89djt, 2024-07-27:
Maybe show the extrusion width colouring? It's taken as read that you've checked the extrusion order (I did too) so just frames of the extrusion widths if they're different might be informative.

agrif, 2024-07-29:
I think I'm seeing exactly the same thing. I have internal threads on my part \([project file zip](https://github.com/user-attachments/files/16409535/nose-cone.zip)\) that are failing to print on PrusaSlicer 2.8.0. I've determined that they are failing because they print the external perimeter of the hole first, and it has no support to stick to and it sags:

[image]

**I believe this is a regression.** In PrusaSlicer 2.7.4, with exactly the same settings, it prints the internal perimeters first, and the print succeeds:

[image]

What I'm seeing agrees with @u89djt's observation that this occurs when internal perimeters merge. This issue might also be the same as the one in #13075.

If I can figure out how to build PrusaSlicer reliably, I will try to bisect and try to figure out which commit changed the order. But maybe somebody familiar with the source will know why this is happening.

agrif, 2024-07-29:
The commit that caused this change is very likely to be [SPE-1963: Improve ordering of perimeters with Arachne perimeter generator][10875082d]. Slicing on this commit prints the external perimeter first, while slicing on the parent commit does the internal perimeter first.

 [10875082d]: https://github.com/prusa3d/PrusaSlicer/commit/10875082de537793f6e62ce719b3e4defd57cfa1

This looks to be completely intentional. I'm mentioning the author @hejllukas to bring this to their attention: It looks like some parts that used to print before this commit now fail or print poorly due to unsupported perimeters. Is there a way to either turn this off in configuration, or to detect this situation and keep the old perimeter order?

eVenom, 2024-07-31:
I think you are into something with SPE-1963 I will try with an older version and get report back.

hejllukas, 2024-08-19:
Hi, unfortunately, this is the side effect of the mitigations for seam gaps in some cases (mentioned here: https://github.com/prusa3d/PrusaSlicer/issues/11914#issuecomment-2182686870).

It is intentional that when Arachne is used on objects with holes and with less than four closed perimeters (not split into several pieces caused by thin parts of objects), then the perimeter of the contour is printed as the last right after the internal perimeter. This, unfortunately, implies that the perimeter of the hole is printed before the internal perimeter.

In the current version, there isn't an option to turn off this behavior. But I admit that, in this case, it likely has a negative impact on overhangs.

u89djt, 2024-08-19:
I'd be inclined to thank them for providing a neat pathological test case.

eVenom, 2024-08-21:
> Hi, unfortunately, this is the side effect of the mitigations for seam gaps in some cases (mentioned here: [#11914 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/11914#issuecomment-2182686870)).
> 
> It is intentional that when Arachne is used on objects with holes and with less than four closed perimeters (not split into several pieces caused by thin parts of objects), then the perimeter of the contour is printed as the last right after the internal perimeter. This, unfortunately, implies that the perimeter of the hole is printed before the internal perimeter.
> 
> In the current version, there isn't an option to turn off this behavior. But I admit that, in this case, it likely has a negative impact on overhangs.

Thank you so much for the reply.  I have found that this example is not so unique, it happens more that expected and yes its a negative impact to the point that it makes the print fail. One thing I don't understand is how reversing the expected wall order resolves the seam gaps.

Thank you again and I am happy that there could be someone possibly working on a solution.

MartijnGevaert, 2024-08-26:
Below is another real world example where this is leading to ugly print artifacts.
In this case, the external perimeter first behavior is caused by a single 2-perimeter wide section.
My printer is underextruding somewhat at the start of travel, I'm still trying to fix that, but this leads to a very visible gap at the seam location.
[image]
[image]

kubispe1, 2024-09-03:
Hi, thanks to everyone for your investigation. As @hejllukas mentioned, this is an intentional change intended to improve seam quality. For now, we will not revert the behavior or add a switch. We have a more general solution and fix planned regarding the Arachne engine.
Sorry for the unpleasant message. Have a nice day

agrif, 2025-03-25:
OrcaSlicer appears to have solved this in SoftFever/OrcaSlicer#7959. I still see the bad behavior in PrusaSlicer 2.9.1.

Any chance you could cross-port the fix, or have an update on the planned Arachne changes?

annevanleyden, 2025-05-18:
I ran into this issue too. I tested this on Cura (also using the Arachne engine), and it too respects your setting for perimeter printing order on inner contours. PrusaSlicer seems to be the only one with this odd behaviour that prevents one from printing nice inner contours if there is an overhang.

agrif, 2026-09-08:
This issue still exists in PrusaSlicer 3.0.0-alpha11. I believe it to be a bug, since it results in a failed print with no way to override it.

It has now been 2 years and one major version bump since @kubispe1's update. Do you have any updates on the planned solution? Is it possible to add a configuration option to restore the old behavior, since the planned solution is either taking longer than expected or does not exist?