prusa3d/PrusaSlicer #13147: Support option that builds off the model

Groggynod3D, 2024-07-24:
Excessive support material waste multiplied by the thousands er perhaps millions of 3D Printers.

This problem could be solved by having support that builds off the model instead of the build plate/print bed.

Obviously this could be done in a hacky way with using support everywhere and paint to remove the 70-90% that isnt needed from the model but would be much easier in reverse order using some sort of angular, arch type, or organic by allowing the user or algorithm to specify or suggest respectively a start point to build from (the root) and end point to support (the branch).

This could "dramastically" reduce print times and filament waste I would venture to say perhaps by an order of magnitude. I know this isnt great for increasing sales for filament manufacturers LOL but it would be amazing for the end user and the planet.

If this has already been suggested or patented I apologize for wasting your time.

u89djt, 2024-07-24:
( a fellow user digging around)
In the normal running of things, supports are placed on the object wherever possible unless we select build plate only.
[image]
Organic supports will not initiate on anything but a horizontal surface because adhesion isn't guaranteed without adhering deeply enough to stop removal without damage. Consider that there is twisting torque and bending leverage on tree-like supports, and these become more significant the more efficient the structure is, so the demands are very different - much more stringent - than on traditional supports, which support vertically upward. When there's enough of a horizontal surface on the object, it might start a support.
Adhesion of efficient branching support structures that initiate on non-horizontal surfaces would require very firm anchors. "Very" is a vague qualifier, but I suppose it's appropriate for a situation where it's hard to predict what loads will be placed.
That problem has been solved for lightning infill, but of course using the same approach externally would mean that the supports aren't removable; certainly not without damaging the object's surface.
The technical barriers to creating supports that vault from inclined surfaces to support higher parts of an object includes finding a way to guarantee sufficient adhesion to support the growing support and then also the exposed volume of the object printed on top of it until it rejoins the rest of the object, while at the same time enabling it to be removed at the end of the print.
With the current organic supports, stability is achieved by trailing down to the bed or another horizontal surface.
Having said all that, we all want quicker, lighter prints. Do you have an example project where object to object jumping supports seem plausible? There's an iceberg of work done already aiming for efficient stable supports (the stuff we see, and the majority that we haven't seen yet). A conspicuous promising example might catch a developer's eye.
We need a really coherent, open wiki where 3d printing concepts, opportunities and progress are categorized and tracked from the public and potential contributors' perspectives. And I wish the code was commented at an enabling quality.

Groggynod3D, 2024-07-24:
You make an excellent point. I had not thought of the aspect of the weight of the support structure and I dont have a specific example although it seems there is already some print stability detection incorporated into the slicing algorithm perhaps that could be used in combination with the support placement. I think it would be more apparent on artistic models as opposed to the more basic geometric ones. All that being said your point definitely makes my idea seem quite more niche

u89djt, 2024-07-24:
You could have a whale of a time concocting elegant helices of support ivy around curved figures. I feel like I've seen that done on a castle somewhere.
Ah. Creases could hide the adhesion of a support root to be broken off. Niches :)

Groggynod3D, 2024-07-24:
Also about the print scarring depending on the polymer a little butane torch does wonders.

u89djt, 2024-07-24:
Aye, and maybe a dremel or usb-powered nail file first so the necessary touch of heat to re-gloss the surface is lighter.

closed 2026-09-29 by prusaslicer-bot[bot]