prusa3d/PrusaSlicer #13148: prusaslicer shows the wrong colors for the multi extruder

ufgum, 2024-07-24:
### Description of the bug

See the images:

[image]

[image]

[image]

### Project file & How to reproduce

just prepare a color print in prusaslicer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

linux

### Printer model

xlis 5 head

Jan-Soustruznik, 2024-08-20:
Hello, @ufgum , could you share your 3MF file, please?  Thank you

ufgum, 2024-08-26:
> Hello, @ufgum , could you share your 3MF file, please? Thank you

Here I created one and made a zip, because 3FM isn't supported here:
[cover filament-left v3.zip](https://github.com/user-attachments/files/16751070/cover.filament-left.v3.zip)

Jan-Soustruznik, 2024-09-06:
@ufgum Yes, now understood. We have there two color Filament color (located: _Filaments > Filament > Color_) and Extruder Color (located: _Printers > Extruder .. > Extruder Color_). For multitool/multimaterial printer are use for paiting visualization the Extruder colors. Filament color is mainly used for single material printers. It's something that's been in the slicer for a long time, and we know it can be confusing. For now it is like this, in the future we will try to unify the colors. I hope that my answer was helpfuly for you and we can close this tiket.

ufgum, 2024-09-10:
> @ufgum Yes, now understood. We have there two color Filament color (located: _Filaments > Filament > Color_) and Extruder Color (located: _Printers > Extruder .. > Extruder Color_). For [...]

It is not confusing, it is just wrong naming or wrong displaying. By the way, it is a great and helpfull view, to see the multicolor Prints, even single color prints in the final colors. So you will be able to swap some colors.

The extruder color is helpfull to slice and edit the print, because the filament color might be too dark to use for editing, but an preview in filament color bevore printing would be great.

But if this is not important for you, you can close the this ticket.

ufgum, 2026-09-19:
This bug still exist