prusa3d/PrusaSlicer #13151: Coloring individual layers

T3chWarri0r, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I want to do multicolor and I can't select specific layers to be painted or done with a different tool head

**Describe the solution you'd like**
I'd love to simply select the layers like when it's already sliced, then select layers and assign tool heads or colors. Then export it into Prusa connect or GCODE. This way also tool head changes could be minimized, because if you paint it by and it's all over the place and the prints ends up to take ages.

**Describe how it would work**
Give the options after slicing to color it this way you could even select infill or supports to be a different tool head 

**Describe alternatives you've considered**
Messing with GCODE, but I'm too new to this and need to get more proficient.

**Additional context**
there should be more tweaking on layer bases.

Jan-Soustruznik, 2024-09-06:
Hello, @T3chWarri0r , I dont know if I understood correctly, but you can use the cut feature and cut the model  into a Parts. Then you can just select cutted part and attach them print tool. 

[image]

T3chWarri0r, 2024-09-06:
I meant more like selecting individual layers to colour them . E.g. I want layer 1-10 blue 11-20 yellow 21-30 green and so forth something like that was what I meant. And this with like how many toolheads or AMS slots you have. —

Peek.2024-09-06.11-04.gif (view on web)

neophyl, 2024-09-06:
Just add a series of height range modifiers and then assign an extruder to each.

[image]

Jan-Soustruznik, 2024-09-09:
Hello, @T3chWarri0r, then you can use method what is mention by @neophyl above, you dont need to cut the model, it will works the same.

rmpel, 2025-12-14:
I am looking for something like this, but manually adding height modifiers would (1) be very labor intensive and (2) not work on part of an object. I would like to see "color mixing" on multicolor painting; choose 2 or more colors and an alternating pattern. For example; I have black in 2 and orange in 3, i choose multi color painting, these 2 colors, and pattern 1:1 each layer, start painting; of the painted part 1 layer is black, one is orange, repeat. With this we could create more colors than filaments available.

closed 2026-09-29 by prusaslicer-bot[bot]