prusa3d/PrusaSlicer #13153: Brimand skirt option for modifacator.

lightingcze, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
Sometimes could be handful to hav the option to turn off on som places the brim for part of the model.
**Describe the solution you'd like**
If will be possible to add modificator with option for brim and skirt. 

Here is our discusion about
https://www.facebook.com/share/p/ZkJLYq6LPdThqCA4/

Snuff1eupagus, 2024-07-25:
@lightingcze Perhaps this will address your need.

https://www.youtube.com/watch?v=dhsbMmosbLU

fe60, 2024-07-25:
Brim blocker have been requested in issue #3554 already.

Snuff1eupagus, 2024-08-12:
@lightingcze You may be in luck, this was just posted.

https://www.youtube.com/watch?v=6CRqNk_JUj4

Hopefully it will help you.

Jan-Soustruznik, 2024-09-09:
closed a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3554

closed 2024-09-09 by Jan-Soustruznik