prusa3d/PrusaSlicer #13154: Crash trying to slice in vase mode after changing nozzle diameter

fionnb, 2024-07-25:
### Description of the bug

To print a vase I put a 0.8mm nozzle into Tool 5 of my Prusa XL.

Then I started PrusaSlicer (v2.8.0-1 on Arch Linux), changed the nozzle diameter from 0.4 to 0.8 on tool 5, adjusted some print settings like extrusion width and clicked slice. It crashes at approx 90% of the progress with the following message on the shell:

`/usr/include/c++/14.1.1/bits/stl_vector.h:1149: std::vector<_Tp, _Alloc>::const_reference std::vector<_Tp, _Alloc>::operator[](size_type) const [with _Tp = double; _Alloc = std::allocator<double>; const_reference = const double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Abort.`

### Project file & How to reproduce

1. Load Project [Spiral_Vase1_p352493.zip](https://github.com/user-attachments/files/16382834/Spiral_Vase1_p352493.zip)
2. Click on Slice
3. Crash before progress = 100%

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-1

### Operating system

Arch Linux

### Printer model

Prusa XL 5tool

Jan-Soustruznik, 2024-08-19:
Hello, @fionnb I cannot reproduce your reported issue. I folow your steps above - Everything bG-code was generated without any problem. Could you more specify the Slicer version  "2.8.0-1" what do you mean this version?

fionnb, 2024-08-23:
2.8.0-1 is the version tag of the Arch Linux package that is currently provided in the official Arch repositories

`$ pacman -Ss prusa-slicer
extra/prusa-slicer 2.8.0-1 [Installed]
    G-code generator for 3D printers (Prusa fork of Slic3r)`

Jan-Soustruznik, 2024-09-06:
Hello @fionnb , I cannot reproduce your reported problem. I used your attached 3mf file, but everything is fine. Could you try it with our Appimage from github(https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0) or from Flathub(https://flathub.org/) example? Maybe the app in official Arch repositories may have some problem .. I dont know.

cc0der, 2024-09-17:
I can reproduce the error on Fedora 40 with PrusaSlicer 2.8.0-3.fc40 from the official Fedora repo.

**Version:** 2.8.0
**Build:** PrusaSlicer-2.8.0+Fedora

Also happens with the Precision cube test model "cube-40-40-40.stl" from [Extrusion multiplier calibration](https://help.prusa3d.com/article/extrusion-multiplier-calibration_2257).

Only happens when printing in spiral vase mode. In normal mode it slices fine.

**printer:** XL 2-tool
**print profile:** `0.20mm SPEED @XLIS 0.4`

fionnb, 2024-09-22:
I upgraded to the official 2.8.1 from Arch/Extra and the crash still happens.

fischer-felix, 2025-04-02:
Is there any update on this? Still happens to me in version 2.9.1 on Arch Linux trying to slice one of [these](https://www.printables.com/model/389267-gridfinity-enhanced-vase-bins-no-label-flaps-90-mo/files) models.

SachCZ, 2025-04-03:
Hi, there is and underlying slicer issue that **did not show up during our testing**. You can read more about it here:
https://github.com/prusa3d/PrusaSlicer/issues/14113#issuecomment-2687398049 (and following the link in the linked comment).

If you want to use a version that we tested, please use the Flatpak version, where we can provide some guarantee.

closed 2026-09-29 by prusaslicer-bot[bot]