prusa3d/PrusaSlicer #13156: Default to gyroid infill for structural presets

rmpratt1, 2024-07-26:
**Is your feature request related to a problem? Please describe.**
The system presets for "STRUCTURAL" prints are intended to provide a default for users to print slower but stronger items. While the grid infill prints reasonably strong the gyroid pattern has increased strength in multiple dimensions. Since these system presets are intended to help new users get started printing quickly and easily the "STRUCTURAL" presets should default to what offers optimal strength. 

**Describe the solution you'd like**
I would like to see the system preset for "STRUCTURAL" type print settings changed to the gyroid infill pattern.

**Describe how it would work**
The print setting options for "STRUCTURAL" type prints is changed to the gyroid infill pattern. 

**Describe alternatives you've considered**
This could be set as an override as suggested in [issue-13131](https://github.com/prusa3d/PrusaSlicer/issues/13131). 

**Additional context**
In my opinion, making this a default is the best option as users who are new to 3D printing are getting default settings that are in line with their expectations.

PaparazziN, 2024-07-26:
Hi @rmpratt1 
I think this issue would be better suited for [this repo](https://github.com/prusa3d/PrusaSlicer-settings-prusa-fff), where printer settings are handled.

Regarding your actual request, there is definitely some merit to using gyroid infill (and is actually my preferred pattern as well) however it would strengthen your case if you added some sources for your claims it is stronger. For example, CNC kitchen found that grid is noticeably stronger than gyroid perpendicularly whilst being slightly weaker than gyroid transverse to the printing direction as seen [here](https://www.youtube.com/watch?v=upELI0HmzHc). However, [Slant3D](https://www.youtube.com/watch?v=IIS9UagZaWI) found that grid is only about 60 % as strong as gyroid in the transverse direction. What I really want to say with this is that the ideal infill is not always clear and depends on your application. 

Another aspect I want to mention when it comes to strength of 3D prints is how the thickness (and amount) of perimeters affect the strength of part and how it plays a significantly larger role than infill pattern according to [CNC Kitchen](https://youtube.com/watch?v=9YaJ0wSKKHA).

u89djt, 2024-07-26:
@PaparazziN You'll see that there are no issues raised publicly in the settings repository you've linked to. They don't solicit inputs there, and they haven't provided a template for submission as they do here, which would certainly be crucial to help users express themselves meaningfully. They're dealing with that internally.
With respect to supporting new users, gyroid infill can fail if filament or printer conditions aren't spot-on. See for example https://github.com/prusa3d/PrusaSlicer/issues/12839
From out here, there's no chance of us really appreciating the breadth of issues prusa have encountered with trying to give new users more technical defaults without extensive practical investigation. Imagine the volume of support calls.
CNC kitchen is providing information for people ready and able to put work into selecting settings that work with their equipment and materials for specific applications. Does he express and justify any wishes to push equipment and software producers to change? The defaults have to cope with users won't necessarily dry their filament or calibrate the equipment as frequently or thoroughly as you.

PaparazziN, 2024-07-26:
@u89djt I am well aware that there are no public issues raised in that repository, which is most likely due to it being quite new as the profile management system changed with the release of 2.8.0. Whilst I do agree it is not entirely clear where discussion of profile preset should be had, to my understanding they have historically been kept in a separate repo such as [this](https://github.com/prusa3d/PrusaSlicer-settings), therefore my recommendation to raise an issue in the previously mentioned repo instead of here.

It also seems that I need to clarify my first response by saying that I believe keeping grid as defaults is most likely a wise decision. Grid infill has been Prusa's recommendation for almost every printed parts for their products, for what I assume being good reasons. I attached those sources to make it bit more nuanced, as it shows that both patterns have their strengths and weaknesses, thus debunking a claim saying one is always better than the other. I mentioned perimeters for a similar reason, as that has a much bigger effect on the strength of a print than which infill pattern you have, thus if you want to make an argument for strength, maybe focus on those parameters instead of infill pattern. 

Whilst being a bit off-topic, I don't really agree with your sentiment about how difficult it is to print with gyroid and I would even argue it is easier to print as it doesn't cross itself like the grid pattern. Gyroid is even the default for the MK3S+ profiles, most likely for good reason. Isn't the point of all form of publications and media to push for change?

u89djt, 2024-07-26:
Have a look at the issue I linked to. I'll leave you to it from here on in.

PaparazziN, 2024-07-26:
@u89djt Can you please explain how the issue you linked is relevant to this issue? I fail to see the relevance of how organic support has anything to do with infill patterns in this context.

u89djt, 2024-07-26:
Yes, it's nothing to do with the organic supports. If you look in the project file they posted and scroll through the tests I carried out, it looks like the gyroid infill has failed to participate effectively in dealing with whatever filament or machine issues they were experiencing.

u89djt, 2024-07-26:
Looking around a bit, consensus seems to be that gyroid infill causes oscillations of the whole printer, which obviously gets worse at higher speeds. My old MK3S still shows gyroid as the default quality infill pattern, but that's generally slower than the speed option, so presumably that reduces the oscillation. XL default profiles don't use gyroid at all, and that agrees with the idea that shaking - and perhaps resonance depending on what the printer is standing on - is a problem with the sustained high frequency sinusoids being driven at the tool head. You can choose to use gyroid, but may be saddled with planting your printer's frame more rigidly.
Maybe input shaping can go haywire if the whole printer is oscillating? CNC's gyroid video was 5 years ago - printing speeds are higher now, and fancier things are happening during extrusion, so I guess it's a different game.

rmpratt1, 2024-07-27:
@u89djt & @PaparazziN, you both bring up many good points. 

TL;DR I think you're both right that it should be left as is (grid infill), but I'll leave this open for a bit pending any additional counter comments.

> (PaparazziN) think this issue would be better suited for [this repo](https://github.com/prusa3d/PrusaSlicer-settings-prusa-fff), where printer settings are handled.

Yeah I'd agree for future issues. For this one specifically, since the discussion has been happening on this thread I'll leave it here and reference it in an issue over in that repo if there ends up being consensus to push for this. 

> (u89djt) From out here, there's no chance of us really appreciating the breadth of issues prusa have encountered with trying to give new users more technical defaults without extensive practical investigation. Imagine the volume of support calls.

This point is what really resonated with me. In my experience, grid infill almost always just works<sup>(TM)</sup>. I think I've had more print failures using gyroid infill, whether that is from gyroid infill iteslf or if it's that I began doing more complicated prints when I switched to more frequently using gyroid is another matter. Either way, one big reason people buy Prusa printers is that they just work<sup>(TM)</sup> so if it decreases that dependability its the wrong choice. 

Interestingly, the only infill type in the [Prusa Knowedge Base article on infill patterns](https://help.prusa3d.com/article/infill-patterns_177130) that mentions potential for print failures is grid. 

_The grid infill is more solid (and has better layer adhesion) than the rectilinear infill, however, it sometimes can cause annoying noise or even a print failure due to the nozzle going over the crossings where material accumulates._

> (PaparazziN) Regarding your actual request, there is definitely some merit to using gyroid infill (and is actually my preferred pattern as well) however it would strengthen your case if you added some sources for your claims it is stronger. 

You hit two of the sources I had. In addition, the [Prusa Knowedge Base article on infill patterns](https://help.prusa3d.com/article/infill-patterns_177130) specifically mentions the strength of gyroid pattern.
_The Gyroid is our favorite and one of the best infills. It’s one of the few 3D structures that provide great support in every direction._
That said I'll look around this weekend and see if I can find any additional sources on gyroid vs grid for strength and print reliability.

u89djt, 2024-07-27:
If we're still rolling, here we go:
If you diff the oldest version of the infill page on waybackmachine with the one up right now, you'll see that the only change in those 2 years and 3 months is correction of typos and removal of "Again" and "However, unlike previously described infills, it". We don't have access to Prusa's perspective except, for example, through noting that the "quality" preset has gone, replaced by "structural". Was that renaming/rebranding chosen because it might be confusing to some users that the gyroid infill previously used in it went away? We don't know. We do know that prusa chose to not use it. So, it's not the manufacturer's default for their printers. I like gyroid too, and it doesn't cause me any problems. But that's me. My perspective on user experience is infinitesimal when compared to what actually happens in the real world. I usually change from grid, too. Every project ends up with different settings all over the place because each project has its own problems and opportunities. A lot of folk want to just print successfully, or maybe tweak from a safe starting point, and that's presumably what the defaults are for. The defaults change with the technology, and also presumably with the size and character of the user base when you're expected to support them.
Is this a github issue? Maybe you'd ask for a feature of returning the old quality preset that's slow enough to prevent gyroid from causing vibration issues. Seems like an edge case that isn't a failure.
Another random thought: what percentage can gyroid go down to without flopping around and failing? That's going to vary significantly with filament quality. You'd have to take responsibility for that - default users probably shouldn't be put in that position? It would be tempting to ask for commentary from prusa, but details of performance like that sound commercially sensitive?
I could be completely wrong. They might be intending to switch to gyroid next week. But that would be based on months of data about customer experience on the other end of the support lines.

rmpratt1, 2024-07-29:
My promised post weekend update…

TL;DR: Identified 40+ sources. Down selected to 15+ relevant articles/videos. General consensus - grid and gyroid infill at the top in terms of strength. Inconsistent testing methods and lack of data ruled out making “apples to apples” comparison. Going to reach out to Prusa team if they have any insights they want to share.

After analyzing more than 15 articles/videos, and sifting through about 25 more that were irrelevant, there were a few patterns appeared in the data. Admittedly, I made a rookie mistake and did not retain my sources and references of which article said what. Instead, I was just keeping a tally of interesting information as it came up and then marked the number of times that was said. I can reconstruct that data from my browsing history, but would be a couple weeks due to commitments the next two weeks. 

For the sake constant terminology:
- **Strength:** A measurement of a particular material and design before it fails due to a stress placed upon the item
- **Fail(ure):** _NOTE: This is inconsistently used across the referenced articles._
  1. The point at which an item begins to show failure due to stress(es) - e.g. deformation lines, discoloration, separation of material.
  2. The point at which an item breaks (in some form defined by the author) due to stress(es) - e.g. cracks formed, snaps in half, crushes
- **Stress:** Is the type(s) of force(s) applied to an object
- **Types of stresses:** _NOTE: stress can be applied across any dimension, e.g. corners, x, y, z_ 
  - **Compressive:** Stress caused by compression or squeezing an object. 
    - _Example:_ The sole of a shoe is experiencing compressive stress while it is being worn
    - _Failure indications:_ Bulges in the uncompressed dimension. Pancaking in the compressed dimension
  - **Tensile:**  Stress caused pulling or elongating an object. 
    - _Example:_ The ropes on a hammock are experiencing tensile stress while it is being sat in
    - _Failure indications:_ Bulging in the elongated dimension. Tearing and/or pancaking in the uncompressed dimension
  - **Sheer:** Stressed caused by applying opposing forces on a dimension
    - _Example:_ A piece of paper experiences sheering stress when it is being cut by scissors 
    - _Failure indications:_ Tearing/breaking/bending of the object, the object being shifted in the dimension to which the shear force is applied

Research findings, I tried to specify the number : 
1. Grid and gyroid infill patterns came out nearly unanimously as the strongest two patterns (15+ vs 1-2 saying something different)
2. Many of these articles (10+) did not specifically call out one as stronger than the other
3. For those that clearly call out specific infills for strength, 3 said grid and 2 gyroid
4. Testing methodologies varied and each seemed to focus on one or two stress terms, thus even for articles providing data or calling out a specific infill type, it is difficult to judge apples to apples
5. Sheering stress seemed to be consistently in favor of gyroid
6. Several articles (3-4) did mention grid being an older pattern. Which leads to it being more tested with more iterative implementation improvements - a.k.a. higher chance for successful print
7. A couple articles (2) suggested one concern for gyroid is that the nature of the pattern causes more vibration, and in some cases harmonic oscillation, which increases the likelihood of failed prints
8. One or two articles stated or implied that gyroid uses less material for the same infill, and posited (admittedly without data) if material volume instead of infill percent was used as the equivalent factor that:
  - Gyroid would have an advantage over grid in sheering and tensile stresses. 
  - Gyroid would be have an advantage over grid, except at the points where grid intersects with the outer surfaces where grid would have the advantage, for compressive stresses. 

Next steps:
- [ ] Garner additional insights and/or suggestions from @u89djt 
- [ ] Garner additional insights and/or suggestions from @PaparazziN 
- [ ] Garner additional insights and/or suggestions from @community
- [ ] Reach out to Prusa product team for comments
- [ ] (maybe) Reach out to project farm to do some testing

u89djt, 2024-07-30:
Briefly, then, since I believe the comments I've made so far do a reasonable job of reflecting my understanding of defaults on commercial products.
You might not be aware that @ community is the whole of gihub, not just this repository.
Making the defaults other than what works best for their user base as established somewhere between general testing (I think you don't have a realistic impression of just how much printing they do) and user support metrics is not going to happen. You're wasting your time here.
Removing friction in replacing the fundamental defaults will leave some users with effectively bricked printers if they mess it up. That's not going to happen.
A better use of your time if you really want the same infill every time would be to learn how to code different defaults for yourself. You will find examples of configurations elsewhere on github, for example.
If you want to really prioritize the strength of your prints, you will chase slicing parameters of every object and part.
OK, so enthusiasm is important, and I've indicated where it should be directed. I'm going to unsubscribe from this post. I think the issue should be closed.
Best wishes
Dave

closed 2025-07-23 by rmpratt1