prusa3d/PrusaSlicer #13161: Prusa XL: Multiple wipe towers - a wipe tower per nozzle diameter

randomev, 2024-07-26:
It's not currently possible to use multiple nozzle diameters (say 0.4, 0.6) with wipe towers in the same print with Prusa XL 5 toolthead version.

PrusaSlicer notifies that wipe tower is not supported in this setup. 

One solution could be to allow multiple wipe towers. One wipe tower where 0.4 tools are wiped and another wipe tower where 0.6 tools would be wiped. This would at least allow to use wipe tower with multiple tools setup.

u89djt, 2024-07-27:
(fellow user who isn't using different nozzle sizes yet) Could you post a zipped 3mf file of an example setup that you want to work? It'll be interesting to see if a workaround can be built without disturbing my defaults. Cheers!

kurtgluck, 2024-08-02:
If you could have multiple wipe towers, where you assigned different sets of extruders to different towers this would also enhance multilateral printing. It would just be easer to have separate PETG and PLA towers than messing with the support structure of the tower and assigning an extruder to its exterior.

LuckyResistor, 2024-09-04:
Adding support for multiple wipe towers would greatly improve print quality and make more complex multi-material prints possible. This would be especially useful for printing with different nozzle sizes or materials that don’t mix well, like PETG and PLA.

**Here’s an example to consider:**

[image]

- **Material Compatibility:** Separate wipe towers would allow materials with different properties, such as PETG and PLA, to be used without issues. When combined in one wipe tower, PETG can get pulled out or dislodged, causing print problems.

- **Better Tower Placement:** As shown in the example, placing different wipe towers on opposite sides (e.g., yellow on the left and black on the right) helps prevent filament strings from crossing into the wrong material or color.

- **Less Color Contamination:** In multicolor prints, having separate wipe towers lowers the risk of color contamination, like stray blobs messing up the print’s look.

- **Improved Recycling:** Keeping colors and materials in their own towers keeps them pure, making them easier to recycle and reuse.

While multiple wipe towers do use more material, the benefits of fewer print failures, less cleanup, and better overall print quality often outweigh the extra cost. In many cases, having a perfect print saves more time and money than the additional material used.

VIN-y, 2024-09-13:
I would like to put in my support for this idea. Nevertheless, I will leave it to the dev to decide to decide on the best way to implement a more sophisticated wipe tower.

My problem with the current prime tower is that I have to chose between multi-material (PETG and PLA) and multi-colour with PETG. As PETG is pron to string, it need to be set as the wipe tower shell ("Wipe tower extruder"), when it is print together with PLA. This also help with keeping the structural integrity of the wipe tower, as the shell is of a single material.

However, this solution does not work so well if you have multiple tool head with PETG, as you can only set 1 to be the ("Wipe tower extruder"). So the others will lack the runway that PETG needed to be stabilised in the nozzle. Furthermore, the prime line (in the centre of the wipe) is still alternating between PLA ans PETG, thus is always at risk of failing to prime correctly. This issue is especially relevance to PETG, due to the risk of blobbing. The PETG blob can cause wipe tower collision with the tool-head and failing (see below)

[image]

beelsebob, 2024-12-13:
Yes please!  I'd really like to be able to use a 0.8mm high flow for infill, while using a 0.25mm for the lovely details on the outside to speed up prints, with current PS, it doesn't work out well.  Another use case - being able to put really fine detailed text on an otherwise functional print.

Graloth, 2025-12-10:
Also putting in my vote for this, as I often print with a mix of abrasive and non-abrasive materials, and would prefer the non-hardened nozzles from scraping over the abrasive material on the wipe tower.

ardenpm, 2026-09-18:
I think this is still relevant since it looks like all of the other variants of "Multiple wipe towers" have been closed as duplicates of this one. While this initially refers to different nozzle diameters, many people are also wanting simply a wipe tower per material to allow incompatible materials to use the wipe towers.

While there were changes to try and "lock" incompatible materials into the wipe tower I don't believe they are completely effective and the simplest solution many have requested is just multiple wipe towers which can be assigned to a given tool (which would cover both the nozzle and material use cases.