prusa3d/PrusaSlicer #13163: Color changing and sequentielle print

ardu1, 2024-07-27:
I have 3 parts on the printbed and each part should have an other color. 

I tried to make a color changing during sequentially printing, but it is not possible.

I want the possibility to insert the M600 like if the sequentially printing is disabled.

I could change the color during printing manually, but it is difficult to find the right moment when the printhead leave the one part and starts the next part to print.
So it would be nice to add this possibility.

u89djt, 2024-07-27:
I think you need to add your filament change code here:
[image]
It looks like your request has been relevant for a while. This isn't the first issue raised, but it gives the solution in the screenshot.
https://github.com/prusa3d/PrusaSlicer/issues/7404

u89djt, 2024-07-27:
Great, let's make sure that gets fed back to the folk on the front line @kubispe1

u89djt, 2024-07-27:
Just noting here that it comes up when you type 'sequential' in the slicer's search box:
[image]

kubispe1, 2024-08-02:
I am closing this issue as a duplicate of #7404 to prevent forking the issues and discussions.

closed 2024-08-02 by kubispe1