prusa3d/PrusaSlicer #13164: More settings for bridges???

Rwide, 2024-07-27:
**Is your feature request related to a problem? Please describe.**
It would be nice with more control over how bridges are made. Sometimes (especially if side of bridge has a slight curve) anchoringpoints fail because there is so little overlap. 

**Describe the solution you'd like**
A setting like infill overlap but for bridges seperately. Also number of perimeters and cholice of pattern would be nice. 

**Describe how it would work**
Not sure how it could work because it seems that Prusaslicer considers skins on top of infill as bridges while considering skins on top of nothing as solid infill??? 

**Describe alternatives you've considered**
When needed, I have to add a height range modifier and add previously mentioned settings to it, but it affects the whole layer and not just the bridges (that Prusaslicer considers to be solid infill).

**Additional context**
[image]
[image]
[image]
[image]

Jan-Soustruznik, 2024-09-09:
Hello, @Rwide could you share a project file, 3MF file, please. Thank you

Rwide, 2024-09-09:
Hello @Jan-Soustruznik It's been a while since I printed this. I made a simple object to better demonstrate the need for these additional bridge settings.
Lets say I wanted to print the object in the pictures with 4 walls or more. It would be a problem. Bridge would fail. In order for a succecful bridge, I could only use 3 walls or less. Therefore, it would be great If it was possible to control number of walls, overlap and pattern for bridge areas. Just remove the ".jpg" from the projectfile to open..
[image]
[image]
[image]
[image]
[image]

Rwide, 2024-09-09:
[Body1.3mf.docx](https://github.com/user-attachments/files/16937264/Body1.3mf.docx)

Jan-Soustruznik, 2024-09-10:
Hello, @Rwide, thank you for your project file. I see that you use the soluble supports. That means that the support have zero distance under the overhangs/bridges so bridges are printed as solid infill - different printing speed,fan speed, extruzion amount etc .. Anyway you have also enabled also "Don’t support bridges" so under "bridges" arent generated supports - so the print will fail. I think that you should to check your print settings first.

Rwide, 2024-09-10:
Ok @Jan-Soustruznik. Thanks! The projectfile is just to demonstrate the issue wich still stands. Just uncheck "generate support material". The bridging area will still fail if using 4 or more walls.

Jan-Soustruznik, 2024-09-12:
Hello, @Rwide , yes that is additional wrong settings ... check the model there is the wall thickness1,2mm, but if you will set 4 or 5 perimeters .. example **0.4**_(Extruzion with for perimeters)_**x4**_(set perimeters)_=**1,6mm** that means - in your model there isn't space for the 4perimeters ... now when you move to the layer where the bridge is generated, solid layer, then 4perimeters is generated, because there is space ... if you want use the 4perimeters or more, you will need first adjust/adapt the model. 
[image]

thisloginalsotaken, 2024-11-21:
> if you want use the 4perimeters or more, you will need first adjust/adapt the model.
Or you can use SuperSlicer, with this option: https://github.com/prusa3d/PrusaSlicer/issues/12721

closed 2026-09-29 by prusaslicer-bot[bot]