prusa3d/PrusaSlicer #13166: Constant "Logged to PrusaAccount as ..." Popup Messages and UI Sluggishness

sgano, 2024-07-27:
### Description of the bug

Every few minutes I am getting a popup window at the bottom of the application that says "Logged to Prusa Account as XXX" (where XXX is my user name)  [see screenshot].   Right before this displays the entire UI freezes and becomes sluggish - which is very disruptive when rotating a model or changing parameter values.  

[image]

### Project file & How to reproduce

Open PrusaSlicer v2.8.0 - login to Prusa Account.   Then just use the application normally and observe popups.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

MacOS 14.5 (intel)

### Printer model

MK4

sgano, 2024-07-27:
Work around:  Log out of account and the issue does not occur.

rconstanzo, 2024-07-28:
Experiencing the same issue.

# Version of PrusaSlicer

Version 2.8.0+MacOS-arm64

# Operating system

macOS Sonoma 14.5 

# Printer model

Prusa mk4

MrVinyla, 2024-07-29:
Same on Windows 11 Pro operating system.

Starlynk1, 2024-08-01:
Mine isn't as bad, but does happen enough to be noticed. Seems that the 'connection check' times out and then when it checks again and succeeds, it pops up this message every time. Really need an option to disable notifications.

rconstanzo, 2024-08-01:
I guess there's a couple parts. One is that checking login status shouldn't interrupt UI/UX under any circumstances, and the second one being why is it checking that often to begin with? It should stay logged in for weeks, if not indefinitely.

VinceKezel, 2024-08-09:
Same issue here. Seeing the logged in notification again every few minutes it seems. I hit the X and it comes back in a few minutes. My internet is solid so I don't think it's a connection issue.

The wording does seem off to me. It says "Logged to Prusa Account as username" when I think it should say "Logged into Prusa account as username"

_PrusaSlicer 2.8.0+win64
Win7
Mini+_

Starlynk1, 2024-08-09:
Quite honestly, there is no need for a notification that you are signed in or not. You can easily see that status in the upper right corner where your username shows. If you get logged out for some reason and try to use an online function, then tell us. No need to say it when we aren't concerned about it.

mdrunk, 2024-08-25:
Same on Windows 11 Pro operating system.
PrusaSlicer Version 2.8.0+win64
Using Google login if that makes any difference.

damccull, 2024-08-26:
Also experiencing this on linux with PrusaSlicer-2.8.0+UNKNOWN. Running NixOS on the unstable channel, though that's unlikely to matter for this issue.

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena