prusa3d/PrusaSlicer #13169: Negative volume cut out

akasus90, 2024-07-28:
Hello there!

**Is your feature request related to a problem? Please describe.**
I'm not good at 3d editing meshes in specific apps, but I feel realy comfortable use Prusa Slicer for this)
Negative volume is my favorite tool in it (MORE HOLES FOR GOD OF HOLES!)), but main problem is that I can save them only in project (3mf) and I like to share my remixes with stl format.

**Describe the solution you'd like**
It will be nice if app-user could not only mark volumes not to print, but edit model it self this way/

**Describe how it would work**
I suggest to add in RMB menu for NV option to cut out this volume in model, make a permanent hole in it))

**Describe alternatives you've considered**
And You could add some tools for modeling, which make this app a slicer with build-in mobel editor!)

**Additional context**
I hope You will consider my reqest.
Have a nice day!
And may the Force be with You!)

mooklepticon, 2024-08-31:
I have the same issue. When searching, I found this: https://github.com/prusa3d/PrusaSlicer/issues/7594

That says:

> Subtraction of negative meshes during export was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

by @lukasmatena 

However, it seems that it's either not actually implemented, or it's not functioning properly, or something else.  It's not working for me either.

Jan-Soustruznik, 2024-09-10:
@mooklepticon @akasus90 Hello, yes, booleans operations to export STL/OBJ feature was implemented in the [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6), but the algorithm have a limitation:  _the algorithm may fail on non-manifold geometries (e.g., those containing self-intersection, etc). In that case, the user is notified, and only positive volumes are exported._  So cannot be applied for all models/cases in this moments. Task created SPE-2480

closed 2025-03-14 by Jan-Soustruznik