prusa3d/PrusaSlicer #13171: No extrusion in first layer error no matter what Z position object is sliced in.

Martin-Danger, 2024-07-28:
### Description of the bug

Silcer returns error "No extrusion in first layer". 
Object is placed on the bed surface so the error is confusing. 
I move the object to below the bed surface, and still the same error occurs. 
Issue arose trying to troubleshoot vase mode printing problem. I changed the object thickness because the 1st layer of the hexagonal cylinder would not print correctly (it kept not printing one or more sides of the hexagon shape but only on the first layer). This was due to object thickness being too close to extrusion width. However, by increasing object thickness, I have fixed one error and the slicer has given me another error. 
Note - Vase mode, 0 bottom layers.
See the 2 attached projects. 
Working as expected: Spool.Spacer.Working.3mf
Not working and returning error: Spool.Spacer.3mf

### Project file & How to reproduce

[Spool Spacer Projects.zip](https://github.com/user-attachments/files/16403122/Spool.Spacer.Projects.zip)
Working model (without the no extrusion in first layer problem) that needed to be modified due to obvious problem.
[image]
Model with no extrusion in first layer error that arose when model thickness was increased.
[image]

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

MacOS 14.5

### Printer model

Prusa i3 mk3.5

u89djt, 2024-07-28:
Hi there, vase mode attempts to lay the faces of the solid it's given in a spiral pattern. You've given it a solid with faces on the outside and the inside, so it can't fit a 0.1 wide spiral for all of them. You need to give it a solid with faces only where you want walls. Here's how yours should look:
[solid Spool Spacer for vase mode.zip](https://github.com/user-attachments/files/16404580/solid.Spool.Spacer.for.vase.mode.zip)
[image]

Martin-Danger, 2024-07-29:
Ok, thanks for that. I wouldn't have thought of that myself. The part I am printing, both surfaces (inside and outside) are functional so I doubt I would have thought of effectively ignoring the inside surface as a workaround.

u89djt, 2024-07-29:
You're very welcome. Can I steer you away from thinking of it as a workaround? Try this: vase mode is a combination of omission of infill and top solid layers, and extruding the walls that would have been printed in non-vase mode, but as a single helix instead of individual layers. There are many situations in which that can't work, plenty that work to make a simple empty vase, and also ways to perform cunning tricks to create light, stable structures such as are common in aircraft printing, for example. Your example here is just one step away from that middle main application because it has no bottom layer. I see you're printing with a 0.8 nozzle - I would be concerned about bed adhesion with a 0.4 nozzle and just a single extrusion.

Martin-Danger, 2024-07-29:
I get you, for sure. I love vase mode and often incorporate it into standard prints by setting bottom layers to a large enough number to print the 'non-vase-mode-printable' portion of the part before switching to vase mode for what can be printed. And with bottom layer adhesion, I will add a bottom layer if adhesion is an issue. In fact, if I did have adhesion problems, I probably would have accidentally found this technique becasue I would have filled in the centre of the part to get a full size 1st layer printed!

closed 2024-07-29 by Martin-Danger

u89djt, 2024-07-29:
Internet (helical) high five :)