prusa3d/PrusaSlicer #13173: Extremely laggy UI in the various configuration menus

EDToaster, 2024-07-28:
### Description of the bug

The various configuration UIs are extremely slow. Changing tabs **within** the Print Settings, Filaments, and Printers tab incurs a ~.5 second delay, and scrolling speed seems to be capped at a few lines a second. This issue is present on versions as far back as 2.6.0 on Windows.

What it feels like is that the app is buffering the scroll wheel inputs, but the UI redraw on each scroll cannot keep up with the rate of the inputs.

<details>
<summary>System Configuration and Details</summary>

```
PrusaSlicer
Version:   2.8.0+win64
Build:     PrusaSlicer-2.8.0+win64-202406270929

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 68,625MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 552.22)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2070 SUPER/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
[... 375 lines cut ...]

### Workarounds 

This issue is present on PrusaSlicer and its derivatives (BambuStudio, OrcaSlicer). A small workaround was found to speed up the UI by about 2x, but is still extremely slow (and potentially dangerous). That workaround is presented here in case it is helpful for triage: https://github.com/SoftFever/OrcaSlicer/issues/3899#issuecomment-2199488175

### Demonstration Video

https://github.com/user-attachments/assets/6595b672-b6f5-4fd1-8611-4ea1e5cf653c

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+win64-202406270929

### Operating system

Windows 11 (build 22631) 64-bit edition

### Printer model

None configured

CerulanLumina, 2024-07-30:
I'm having this issue on Linux-

- **OS**: Arch Linux x86_64 rolling
- **Kernel**: 6.10.2-zen1-1-zen
- **DE**: Plasma 6.1.3 **_Wayland_**
- **WM**: KWin
- **GPU**: AMD Radeon RX 6700 XT

Other software based on Slic3r, including Bambu Studio and OrcaSlicer are suffering the same issue, for me at least.

Interestingly, this also affects other apps using GTK. It does not affect apps running natively on Wayland (only XWayland compat), though I can't test slicers since they do not support Wayland.

It may not be the same issue since it appeared after a system update, which included prusa slicer -> 2.8.0, but downgrading to 2.7 doesn't resolve it.

tealblu, 2024-09-07:
Also having this issue on linux.

EndeavourOS
CachyOS default kernel
Hyprland (wayland)
AMD Ryzen CPU
AMD Rx 6700 XT

Like the above commenter, Orca also completely breaks all GTK applications. That said, sometimes Orca works fine, sometimes it's unusable. More often the latter.

LuminaSapphira, 2024-09-07:
For anyone experiencing this issue on Linux, it appears to be a kernel regression with AMD GPUs, and therefore probably not related to the Windows report from OP. See the Gitlab for more details: https://gitlab.freedesktop.org/drm/amd/-/issues/3528#note_2555321

Felix14-v2, 2024-11-13:
> probably not related to the Windows report from OP.

I'm also experiencing the same issue with the laggy scrolling and the extremely slow UI on Windows 11, with both Intel and Nvidia GPUs.

legend069, 2024-12-01:
I've found the same issue but with the object list, when the object list is full and it gets a scroll bar.

EDToaster, 2025-03-08:
The UI when installing through WSL/Ubuntu does not run in to the same issue.

Felix14-v2, 2026-09-18:
The new UI doesn't feel so laggy, but it consumes 100% GPU even when moving a mouse over it, which is a shame.