prusa3d/PrusaSlicer #13174: Per-surface ironing with selection & optional non-planar ironing

k1-801, 2024-07-29:
**Is your feature request related to a problem? Please describe.**
Sometimes ironing is only needed on a particular surface. Adding modifiers to achieve the desired result can be tricky, and if ironing would work best with different direction angles, it becomes impossible, as it currently only works along the X axis regardless of everything else.

**Describe the solution you'd like**
An additional tool on the left panel called "ironing" that opens a surface selector that allows to choose a top surface to work with (in a similar fashion to how ruler tool surface selection works, but only allowing surfaces to be selected if the angle to a vertical is greater than 60 degrees, i.e. if there is at least some amount of surface that can actually be ironed). Each selected surface would add a separate ironing "modifier" to the object that only describes ironing parameters for this particular surface, and the surface can not be re-selected. In the MVP implementation those parameters would only include the surface itself and a line (or an angle) to adjust the ironing direction. This way ironing would also be visible in table preview before slicing is done. This would be a tool independent from the original ironing settings, as "ironing all layers" is still a needed feature for transparent prints, and it is impossible to select internal layers to be ironed with the proposed solution - also backward compatibility with old project files is desired.

**Additional context**
An extended version of this tool could allow some form of non-planar ironing, where the nozzle would be allowed to move vertically to glide on the surface, effectively smoothening the layer lines.

This would involve settings, similar to ones required for sequential printing - i.e., a maximum height for non-planar ironing (initially set to 0 to disable the feature, making it 100% planar, a typical reasonable value could be between 2 and 12 millimeters depending on the nozzle and hotend setup), a maximum angle (also limited by the nozzle shape).
It would also require the extruder settings to contain the outer diameter of the nozzle tip to ensure the correct offsets when ironing a slope, not to graze the printed surface too deep.
For surfaces where the height difference surpasses the height limit set for non-planar ironing, the best way would be to perform it in steps (i.e. print the first N layers of the surface, iron them, then the next N and iron only those, potentially allowing a small overlap).

There are two distinct scenarios where it could be used: ironing a flat slope and ironing a curved surface.
1. Ironing a flat slope would work in exactly the same way as described above, but only performing the ironing after the last layer of the slope has been printed, and working along the whole slope, top-to-bottom. 
2. Ironing a curved surface is trickier. For this, the surface selection would work closer to how "smart fill" works in multimaterial painting, possibly with actual "paint-on" ironing selection. Due to the tricky nature of following a curved surface with a nozzle, I suggest adding a few presets for how the toolpath is expected to go on these surfaces:
* spiral (chooses a peak and works it's way down until it meets a local minimum, then the next peak, and so on) - fast, but may leave marks similar to layer lines
* radial (chooses a peak, works it's way straight to a local minimum, then shifts too the side and glides back to the peak, when it works through 360 degrees around it it shifts to the next peak)
* trace (moves parallel to a single line like classic ironing, but simultaneously moving up and down to trace the surface).

Non-planar ironing could resolve 95% of what is normally required from non-planar slicing, without making full non-planar layers.

u89djt, 2024-07-31:
Existing request: https://github.com/prusa3d/PrusaSlicer/issues/2794
Briefly from a fellow user: ironing only works in the XY plane with a vertically mounted nozzle. This is because the flat, horizontal surface of the nozzle is used. A small amount of filament is extruded during ironing for at least two purposes: to fill the troughs between the target extrusions below, and to lubricate the progress of the nozzle with already-melted material while the upper regions of the target extrusions soften and move.
If you try to iron anything off-plane, you'll end up with some combination of piled-up plastic ahead and either side of ironing moves in +Z, and a channel gouged behind an ironing move in -Z.
If you happen to have a tool that can tip round X and Y (I've never met one, but I see there are experiments in that direction to enable steeper non-planar printing) then you could theoretically revisit an out-of-plane surface. You would have to establish that the tool won't crash into surrounding structures on the bed.

k1-801, 2024-07-31:
> If you try to iron anything off-plane, you'll end up with some combination of piled-up plastic ahead and either side of ironing moves in +Z, and a channel gouged behind an ironing move in -Z.

Which would only happen if we asssume the point of contact of the nozzle with the part is the center of the nozzle opening. I specifically mentioned that the outer diameter of the nozzle tip would also have to be configured. This way, if we assume that the point of contact is located at the same height, but on the outer rim of the nozzle tip, on the downward movement it would indeed create a shallow groove, but it's shape would be a conical cross-section (not exactly a circle, and if the line shift for ironing is small enough (and it is), the edges of the groove would be raised by only a negligible amount, that can be compensated for by the following upward motion with no extrusion. On the other hand, in some cases it would be more beneficial to use only the downward motion with constant extrusion, for these scenarios it's best to use the helical mode described above - it was mentioned that it only goes down from peaks (works well so long as it's possible to do full revolutions around the "cone") - however, doing so may kill the effect of the ironing.

If you wonder what kind of an offset I'm proposing, here's how I see it: We take the point on the surface where we assume we are supposed to go, and then we build a vector perpendicular to that spot on the surface. If the X and Y coordinates are zero (surface is horizontal), we stay where we are, otherwise we discard the Z coordinate in the vector and make it's length equal to the outer radius of the nozzle tip. This way, if there is any slope on that surface, we now only touch it with the outer edge of the nozzle.
Then, it may be a good idea to increase flowrate if the actual plastic surface is not on the same height as we are now at. We know that tge surface is close to horizontal, so with layer height of 0.2, half of the time we are moving we will be more than 0.1mm away from actual plastic layer below us, so the flow needs to be increased, then slowly decreased as we go down and get closer to the plastic, so we hit the configured value (10-15%) when we touch the next layer. I apologize for not providing actual formula for the flow.
On the way up, we will not be able to flatten the plastic if we extrude any, so we either move with no extrusion, or we go up to the next starting point immediately and we only iron downward.

u89djt, 2024-07-31:
I look forward to a demonstration. Pulsing the extrusion to coincide exactly with the complex shapes of the troughs between perimeters will be impressive.

k1-801, 2024-08-01:
> I look forward to a demonstration. Pulsing the extrusion to coincide exactly with the complex shapes of the troughs between perimeters will be impressive.

Well, if you really want me to implement the whole thing on my own, I can give it a try, but it will take months probably, as I am not familiar with the slicer codebase =\

For now, I'd stick with the MVP - just being able to select surfaces that actually need to be ironed would be a massive improvement.

Tupson444, 2026-09-19:
still relevant