prusa3d/PrusaSlicer #13176: 2.8 Colour change bar UI needs to show colour changes above the current layer marker - now it's just black.

prowlmedia, 2024-07-29:
2.8 version has changed the he colour change layer bar:
You now cannot see the colour changes above the current layer selected -  it just shows as black.  Before it showed ALL the colours.   

You can only see all the colours if the current selected layer is at the top.  

The orange lines / ticks that are now created on the bar at colour changed are almost invisible.   I. am guessing they are only 1 Pix lines - they need to be a lot thicker. 

[image]

Jan-Soustruznik, 2024-09-17:
Hello, @prowlmedia, could you more specify why it is necessary to have see all color.

prowlmedia, 2024-09-17:
So I can see the changes visually properly.  Like I have always done.  It’s just unintuitive.  The old full bar with numbered layer ticks was much more useful.

prowlmedia, 2024-10-16:
I feel as thought it should be an option to have the Ticks, layer height / number / changes and  and the colours all on screen at once.  

It's very unintuitive to have to mouse over the ultra thin change lines to know what layer 
Would also be incredibly useful to be able to DRAG those change lines up and down instead of having to delete and add a new change,  because it assigns a random colour each time that you have to go in and change.

closed 2026-09-29 by prusaslicer-bot[bot]