prusa3d/PrusaSlicer #13180: Raft Update for 2 Layer and Soluble Filament

Starlynk1, 2024-07-30:
**Is your feature request related to a problem? Please describe.**
Using Soluble filament like PLA, it is hard to get a good first layer adhesion due to the way the material is. 

**Describe the solution you'd like**
Currently I set up Prusaslicer to make a raft with a top layer interface that is soluble and the first layer is not. While this works, it doesn't allow me to use a full support. In order to make this work, it needs to be an interface support. That is not optimal in many situations where I need the supports to dissolve in hard to reach places.

**Describe how it would work**
First off, Raft should be seperated from the 'Support Material/raft/skirt extruder' and from the 'Support Material/raft interface' section of the Multiple Extruders. Allowing Raft to have its own paramaters for what extruder I think would be the easiest path. This way Supports can still be set up as full support.

In the Support Material tab, Raft section should have it's own options for Options for support material. 

**Describe alternatives you've considered**
I have stated how I do a limited workaround, but that still does not allow me to do full supports

**Additional context**

u89djt, 2024-07-31:
Fellow user who enjoys finding workarounds: if you can post a 3mf of a project that would like to Just Work, odds are someone here will come up with something that achieves what you need in the meantime.

Starlynk1, 2024-07-31:
Attached are two examples of what I am able to do. I have been able to make a 2 layer raft with soluble on the top layer, which for the example seems to work for the most part because this specific model does not need supports.

Second example is doing a full support with 2 raft layers, unfortunately though, I can't make first layer as not soluble, its either all soluble or use interface, which for a model like this with hidden areas, it would be difficult to remove the non-soluble supports.

Preferably would like to see an option or find a way to do normal full supports with a raft, but also be able to set the first raft layer to non-soluble.
[Uploading No Support with Raft example.zip…]()

u89djt, 2024-08-01:
OK, and what would you prefer to be able to do? What do you actually need to support? I see that there are some bridge extrusions within the joints. You would benefit if someone can come up with a solution you haven't thought of, so your actual goal rather than what you've tried is important.

u89djt, 2024-08-01:
Would something like this do? 
[hack Full Support with Raft example.zip](https://github.com/user-attachments/files/16454331/hack.Full.Support.with.Raft.example.zip)
[image]
[image]

Starlynk1, 2024-08-01:
The goal is to have full soluble supports and a 2 layer raft where the bottom layer is not soluble.

image.png (view on web)<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_user-2Dattachments_assets_abd87750-2D01dd-2D4772-2D8d67-2Db7f4f3619aca&d=DwMFaQ&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=XodWtLBkFpvdbc-aEppP_ibPteZq0Cpnye3A6MMh-wk&m=V5i9769IaXmQOoakTeF9ZBNmEZGDgs8Nk-kEiGiNpVYZR9Rs8oEHSqZdTEZK6eAF&s=aCfISqIg5LTQCq_68g03rQ8SEGFrsK2RqXi1eY1mP8s&e=>
image.png (view on web)<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_user-2Dattachments_assets_ce40d10e-2D0946-2D47ff-2D9cf8-2Dc5d1884f49a0&d=DwMFaQ&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=XodWtLBkFpvdbc-aEppP_ibPteZq0Cpnye3A6MMh-wk&m=V5i9769IaXmQOoakTeF9ZBNmEZGDgs8Nk-kEiGiNpVYZR9Rs8oEHSqZdTEZK6eAF&s=EghsaUt-B6V0Q1OxCu5ZmSsnWnm2ciOM0aizTrXjw3Y&e=>

u89djt, 2024-08-01:
Cool, that's effectively what I've done in the 3mf file above. I absolutely agree that it would be great to be able to do it with intrinsic parameter settings.

closed 2026-09-29 by prusaslicer-bot[bot]