prusa3d/PrusaSlicer #13181: Editing file name when sending to Prusa Connect for SL1s causes file extension match error

skellatore, 2024-07-31:
### Description of the bug

When sending a file to a SL1s Printer via PrusaConnect, if you edit the file-name, it results in a error message saying:
"File extension can be one of .gcode, .gco, .g, .bgc, .bgcode, .sl1, .sl1s, .m1, .bbf". This is despite the file extension being correctly set. Changing the file-extension to be ".sl" clears the error but then obviously fails because the extension is wrong. Removing the extension altogether also has same effect.

If the name is not changed/edited in any way, the error does not exist.

### Project file & How to reproduce

1. Slice a file for a SL1s using PS 2.8.0
2. click "Send to Connect" button
3. edit the file name - even delete one character from the beginning without changing the file extension at the end
4. error appears and the confirm button is greyed out.

Screenshot showing unedited file name with correct file extension and no error:
[image]

Screenshot showing edited file name with correct file extension and error:
[image]

Screenshot showing edited filename with no file extension and no error:
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11

### Printer model

SL1s 1.8.0-beta.0

myParakeet, 2024-07-31:
Related to #13184 (happening for multiple printer types)

Jan-Soustruznik, 2024-09-13:
Hello, @skellatore, @myParakeet , could you try with latest release of PrusaSlicer 2.8.1-rc1 released yesterday (https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1). We did together with connect Team an enhancement of communication Slicer <> Connect. It should be Ok. Now I cant reproduce the reported issue above.

closed 2026-09-29 by prusaslicer-bot[bot]