prusa3d/PrusaSlicer #13182: Sending print to SL1s marked with "set ready and print now" via PrusaConnect results in error

skellatore, 2024-07-31:
### Description of the bug

When sending a file to a SL1s via Prusa Connect and PrusaSlicer 2.8.0+win64 utilising the "Set ready and print now" option, the file uploads correctly but then an error message is presented and the print is not started.

Error uploading to print host:
HTTP 409: {"message": "Print start of the downloaded file from the print queue is not possible due to invalid printer state", "code": "CONFLICT_AFTER_DOWNLOAD_PRINT"}

### Project file & How to reproduce

1. Slice a file for a SL1s
2. Send file to the SL1s via Prusa Connect with the  "Set ready and print now"
[image]
3. File uploads correctly, but an error message is presented and print is not started
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11

### Printer model

SL1s 1.8.0-beta.0

GuruChoc, 2025-04-02:
Getting the same thing just now with 3.5S and connect. Printer is showing offline when it is online.

Error uploading to print host:
HTTP 409: {"message": "Print start of the downloaded file from the print queue is not possible due to invalid printer state", "code": "CONFLICT_AFTER_DOWNLOAD_PRINT"}

arr2036, 2025-04-02:
Same issue with a MK4S.

Jack-The-Front, 2025-05-01:
getting same issue with PS 2.9.2 and MK4S, usually with larger prints (15mb+)

[image]

wieli99, 2025-05-02:
Try upgrading your Printer Firmware, fixed it for me (MK4S, upgraded to 6.2.4)

closed 2026-09-29 by prusaslicer-bot[bot]