prusa3d/PrusaSlicer #13183: OBJ Toolpath Export Missing Ironing

PilotMatt, 2024-07-31:
### Description of the bug

Using "Iron Topmost Layer" in the slicer and "Export Toolpaths to OBJ" ironing is never rendered out to the OBJ.  

### Project file & How to reproduce

Slice a file w/ Ironing on .  
File -> Export -> Export toolpaths to OBJ.
RESULT: No ironing on OBJ.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Win 11

### Printer model

Custom Printer (used for Rendering without the Purge line)

u89djt, 2024-07-31:
It turns out the ironing paths are in the obj, but they're half a layer height too low. If you increase ironing spacing to 0.5mm, you'll see where they are because they don't merge into a confusing single layer that's hard to recognize - they're separate lines that you can see between the main extrusions:
[image]
Here's a comparison of where they're shown in the slicer and where they appear in the obj:
[image]
[image]
And the project file and obj shown there:
[ironingpathshalfheighttoolow.zip](https://github.com/user-attachments/files/16440534/ironingpathshalfheighttoolow.zip)
I used layer heights 0.15 and 0.3 to test whether the half height difference is consistent.

PilotMatt, 2024-08-01:
awesome thank you!  Is there a way to generate the toolpath OBJ from the
CLI?

u89djt, 2024-08-01:
(I'm just another user) there's an open issue indicating that you can't get paths out via the CLI, but I haven't tried it https://github.com/prusa3d/PrusaSlicer/issues/5076
In fact we seem to have a messy luxury - here's an older piece mentioning that the triangles are taken from the graphics card's data https://github.com/prusa3d/PrusaSlicer/issues/3046 and if that still pertains, we won't get anything via the command line.
I think the ironing paths appearing in the centre of the previous extrusion is a mistake/bug, since it appears in the right place in the on-screen graphics.
If you can share what you want to do with the data, there are other folk following the repository issues who might know something helpful.

PilotMatt, 2024-08-01:
I'm trying to use blender to render the OBJ in a script.  No matter what I do, line spacing, setting iron flow to 100%, the ironing lines don't seem to show up.  Attached render where the ironing flow is set to 100%.  It's very visible in the gcode viewer.  And I can see the ironing lines when I import it into Fusion.  But they get lost in Blender.

[image]

u89djt, 2024-08-02:
Oh, bespoke product preview? Nice. I would expect prusa to enjoy a valuable application for the export function.
So it's "just" a matter of adding half the layer height to the z coordinate of the triangles of the ironing pass. They're treated as part of the layer they're added to in the preview system so I guess that offset was lost in translation somewhere. If you iron it two different percentages and perform a diff, you'll presumably see which are the ironing triangles. If you're using a uniform layer height, that's your workaround for now, I guess.

PilotMatt, 2024-08-16:
Yes, it's something like that! :) https://auntlosupplyco.etsy.com

u89djt, 2024-08-16:
Good stuff. If you put together a pipeline for fixing the ironing location, you could post it to printables?

PilotMatt, 2024-08-17:
I didn't didn't actually fix anything  My workaround is just print config with very narrow top infill lines!

u89djt, 2024-08-17:
Heh, this is the way.

closed 2026-09-29 by prusaslicer-bot[bot]