prusa3d/PrusaSlicer #13184: User edited Send to Connect name doesn't recognize extensions correctly

myParakeet, 2024-07-31:
### Description of the bug

If I edit the name in the "Schedule Print" dialog (after clicking my printer from the Send to Connect button), it the confirm button is greyed out even with the correct .bgcode extension. If it edit it to .bgcod then the button becomes clickable, but results in an error. 
[image]

### Project file & How to reproduce

I am not sure if I am allowed to upload the file. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

windows 10

### Printer model

Prusa XL

AndrewFarley, 2024-07-31:
Same issue here.  :(  Can't customize name, and without a customized name sometime get name conflicts.  :(

Jan-Soustruznik, 2024-09-13:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13181

closed 2024-09-13 by Jan-Soustruznik